tzapu/WiFiManager #555: Should captive portal screen close automatically upon connection to WiFi?

chrisallick, 2018-03-14:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			ESP12 (esp8266, and specifically the Huzzah from Aadafruit)

Core Version:      	2.4.1

### Description

Upon entering my WiFi credentials, and successfully connecting, the captive portal doesn't close. Is there a way to make it auto close? I suppose it could be solved with a message, but would be a better experience if it closed.

Also, in case anyone gets to this message and cannot get your ESP8266 to connect to wifi or even open the captive portal. Master never worked for me. Failed to ever open the portal or allow my phone to connect to the AP. Development branch worked immediately as well as the Ken Taylor fork master branch.

Additional libraries: 

### Sketch

```
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("SweetAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :) _______");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}

```

### Debug Messages

```
*WM: Connect to new AP [SUCCESS] 
*WM: Got IP Address: 
*WM: 172.20.22.171 
*WM: disconnect configportal 
*WM: disconnect configportal - softAPdisconnect FAILED 
*WM: config portal exiting 
connected...yeey :) _______
*WM: unloading 
```

tablatronix, 2018-03-14:
https://github.com/esp8266/Arduino/issues/3793
This was supposed to be fixed, I fixed it.

hmm

chrisallick, 2018-03-14:
*shrug* let me know if there is anything else I can tell you about my setup or debug messages I can share.

tablatronix, 2018-03-14:
I was testing something and must have commited it

closed 2018-03-14 by tablatronix

chrisallick, 2018-03-14:
Oh well now that I see your commit I'm looking at WiFiManager.cpp and seeing that that
`// ret = WiFi.softAPdisconnect(false);`

tablatronix, 2018-03-14:
yeah , even funnier that I spent 20 minutes debugging it.

chrisallick, 2018-03-14:
Let me test it real quick, brb.

chrisallick, 2018-03-14:
It worked! I suppose people might want to call that manually so I could understand why it would be a setting or a function call rather than a default behavior.

commit 29edaed0 (2018-03-14): fixes #555 oops