tzapu/WiFiManager #566: wifi connection timeout

mutthunaveen, 2018-03-17:
Hello

im using your code for configuring and connecting wifi. it is great.
im in a environment where there will be frequent power fluctuations. so, each time my wifi router will take around 60 sec to init. by the time your code will goto AP mode and ask for config change.

after searching few threads in this repository i found this.. ->  
`wifiManager.setConnectTimeout(70);`  i used 70sec as timeout.

above code as shown below.. 
`  wifiManager.setConnectTimeout(70); // seconds wait before AP
  if (!wifiManager.autoConnect()) {`

after compiling the esp, i restarted both wifi router and ESP,

> Wifi router in active at 50sec

> ESP waits for 70 sec and then finally opens AP.. 

any Help??

thank you.

tablatronix, 2018-03-17:
Connect timeout is how long to leave the portal open before it closes.. not sure what you are trying to do. You might want to just add your own connected wait loop and only start autoconnect if not connected after that interval, so it doesnt always start the configportal.

mutthunaveen, 2018-03-18:
Ok... Got it... I need to develop my own code for connecting before it goes to autoconnect.....

I will explain again my problem....Here is my problem for which i need a solution....

I have a Wi-Fi router in my home... And my Wi-Fi router is prone to several restarts due to power fluctuations. Each time a power on reset causes 60 seconds for my Wi-Fi router to switch on completely.. after 60 second any device can be connected without any hassle... Before 60 seconds I cannot make because Wi-Fi router will be in initiation phase..
So when I program my ESP using Wi-Fi manager, during power on reset my ESP checks for available networks and if you did not find anything it immediately goes to AP mode asking configuration settings.. 

In my case my ESP should continuosly try to connect until 60 seconds, with my predefined credentials available in ESP. if predefined credential Wi-Fi network is not available even after 60 seconds then the ESP is allowed to go to auto connect on AP mode..

I hope you understand my problem... Thanks for the help

tablatronix, 2018-03-18:
Nono the esp autoconnects on its own, you can just add a small while not connected loop before autoconnect

malikit786, 2020-12-16:
in my cases wifiManager.setConnectTimeout(180); does not wait for 180 sec, But only 60 sec. my router takes about 2 minutes to setup after power outage my esp does not connect to the router until it hard reset.
code is here for your reviews.

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
    //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //reset settings - for testing
  //wifiManager.resetSettings();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds

//******this
   wifiManager.setConnectTimeout(180);

  wifiManager.setTimeout(180);
  
  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  } 

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
 
}

void loop() {
  // put your main code here, to run repeatedly:

}

tablatronix, 2020-12-16:
hmm interesting, not sure why the timer would be wrong

tarontop, 2021-03-17:
> 在我的情况下wifiManager.setConnectTimeout（180）; 不等待180秒，而仅等待60秒。停电后，路由器大约需要2分钟的设置时间，直到硬重置，我的esp才连接到路由器。
> 代码在这里供您查看。
> 
> #include <ESP8266WiFi.h> // https://github.com/esp8266/Arduino
> //需要库
> #include <DNSServer.h>
> #include <ESP8266WebServer.h>
> #include <WiFiManager.h> // [https： //github.com/tzapu/WiFiManager](https://github.com/tzapu/WiFiManager)
> 
> void setup（）{
> //将您的安装代码放在此处，以运行一次：
> Serial.begin（115200）;
> // WiFiManager
> //本地初始化。业务完成后，无需将其保留在
> WiFiManager周围。
> //重置设置-用于测试
> //wifiManager.resetSettings（）;
> 
> //设置超时，直到关闭配置门户
> //有用以使其全部重试或进入睡眠状态
> //只需几秒钟
> 
> // ****** this
> wifiManager.setConnectTimeout（180）;
> 
> wifiManager.setTimeout（180）;
> 
> //获取ssid并通过并尝试连接
> //如果未连接则启动具有指定名称的访问点
> //此处为“ AutoConnectAP”
> //并进入阻塞循环等待配置
> if（！wifiManager.autoConnect（“ AutoConnectAP“））{
> Serial.println（”连接失败并超时“）;
> 延迟（3000）;
> //重置并重试，或者将其置于深度睡眠状态
> ESP.reset（）;
> 延迟（5000）;
> }
> 
> //如果到达这里，则说明您已连接到WiFi
> Serial.println（“ connected ... yeey :)”）;
> 
> }
> 
> void loop（）{
> //将您的主要代码放在此处，以重复运行：
> 
> }

Yes, I also have this kind of problem. Sometimes this timer is a random one. I don’t know if you have a solution.