tzapu/WiFiManager #573: Parameter ID's with '_' are no longer accepted?

liebman, 2018-03-21:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32:   Esp8266

Hardware:			ESP-12e

Core Version:      	staging

### Description

I attempted to move a project with 25+ parameters to the development version and see that adding parameters with IDs with '_' are no longer accepted.  Is this intentional?

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp
#include "Arduino.h"
#include "WiFiManager.h"

void setup()
{
    Serial.begin(115200);
    Serial.println("creating WiFiManager");
    WiFiManager wm;
    Serial.println("adding parameters");
    WiFiManagerParameter ntp_server( "ntp_server", "NTP Server", "0.zoddotcom.pool.ntp.org", 32);
    wm.addParameter(    &ntp_server);
    WiFiManagerParameter tc1_label("<p>1st Time Change</p>");
    wm.addParameter(    &tc1_label);
    WiFiManagerParameter tc1_occurence("tc1_occurrence", "occurrence", "2", 3);
    wm.addParameter(    &tc1_occurence);
    WiFiManagerParameter tc1_day_of_week("tc1_day_of_week", "Day of Week (Sun=0)", "0", 2);
    wm.addParameter(    &tc1_day_of_week);
    WiFiManagerParameter tc1_day_offset("tc1_day_offset", "day offset", "0", 3);
    wm.addParameter(    &tc1_day_offset);
    WiFiManagerParameter tc1_month("tc1_month", "Month (Jan=1)", "3", 3);
    wm.addParameter(    &tc1_month);
    WiFiManagerParameter tc1_hour("tc1_hour", "Hour (0-23)", "2", 3);
    wm.addParameter(    &tc1_hour);
    WiFiManagerParameter tc1_offset("tc1_offset", "Time Offset", "-25200", 8);
    wm.addParameter(    &tc1_offset);
    WiFiManagerParameter tc2_label("<p>2nd Time Change</p>");
    wm.addParameter(    &tc2_label);
    WiFiManagerParameter tc2_occurence("tc2_occurrence", "occurrence", "1", 2);
    wm.addParameter(    &tc2_occurence);
    WiFiManagerParameter tc2_day_of_week("tc2_day_of_week", "Day of Week (Sun=0)", "0", 2);
    wm.addParameter(    &tc2_day_of_week);
    WiFiManagerParameter tc2_day_offset("tc2_day_offset", "day offset", "0", 3);
    wm.addParameter(    &tc2_day_offset);
    WiFiManagerParameter tc2_month("tc2_month", "Month (Jan=1)", "11", 3);
    wm.addParameter(    &tc2_month);
    WiFiManagerParameter tc2_hour("tc2_hour", "Hour (0-23)", "2", 3);
    wm.addParameter(    &tc2_hour);
    WiFiManagerParameter tc2_offset("tc2_offset", "Time Offset", "-28800", 8);
    wm.addParameter(    &tc2_offset);
    WiFiManagerParameter advance_label("<p>Advanced Settings!</p>");
    wm.addParameter(    &advance_label);
    WiFiManagerParameter no_sleep("stay_awake", "Stay Awake 'true'", "", 8);
    wm.addParameter(    &no_sleep);
    WiFiManagerParameter sleep_duration("sleep_duration", "Sleep", "3600", 8);
    wm.addParameter(    &sleep_duration);
    WiFiManagerParameter tp_duration("tp_duration", "Tick Pulse", "32", 8);
    wm.addParameter(    &tp_duration);
    WiFiManagerParameter tp_duty("tp_duty", "Tick Pulse Duty", "43", 8);
    wm.addParameter(    &tp_duty);
    WiFiManagerParameter ap_start("ap_start", "Adjust Start Pulse", "34", 4);
    wm.addParameter(    &ap_start);
    WiFiManagerParameter ap_duration("ap_duration", "Adjust Pulse", "17", 4);
    wm.addParameter(    &ap_duration);
    WiFiManagerParameter ap_duty("ap_duty", "Adjust Pulse Duty", "45", 8);
    wm.addParameter(    &ap_duty);
    WiFiManagerParameter ap_delay("ap_delay", "Adjust Delay", "9", 4);
    wm.addParameter(    &ap_delay);
    WiFiManagerParameter network_logger_host("network_logger_host", "Network Log Host", "", 32);
    wm.addParameter(    &network_logger_host);
    WiFiManagerParameter network_logger_port("network_logger_port", "Network Log Port", "", 6);
    wm.addParameter(    &network_logger_port);
    WiFiManagerParameter clear_ntp_persist("clear_ntp_persist", "Clear NTP Persist 'true'", "", 8);
    wm.addParameter(    &clear_ntp_persist);
    Serial.println("connecting");
    wm.autoConnect();
    Serial.println("done");
}

void loop()
{
    delay(1000);
}

```

### Debug Messages

```
SDK:2.2.1(cfd48f3)/Core:unspecified/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
creating WiFiManager
adding parameters
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: Added Parameter:
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: Added Parameter:
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: Added Parameter:
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
*WM: [ERROR] parameter IDs can only contain alpha numeric chars 
connecting
*WM: AutoConnect 
mode : sta(18:fe:34:cd:a6:24) + softAP(1a:fe:34:cd:a6:24)
add if0
*WM: Connecting as wifi client... 
*WM: No saved credentials, skipping wifi 
*WM: Connection result: WL_NO_SSID_AVAIL
del if0
mode : softAP(1a:fe:34:cd:a6:24)
*WM: Disabling STA 
*WM: Enabling AP 
*WM: StartAP with SSID:  ESP_13477412
*WM: AP has anonymous access! 
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: AP IP address: 192.168.4.1
*WM: Starting Web Portal 
*WM: HTTP server started 

```

tablatronix, 2018-03-21:
let me find a better fix, that is not correct

closed 2018-03-22 by tablatronix

woodenplastic, 2023-07-04:
Hi, i have the same problem. I use the newest version and cant give parameter id. i tried everything, including naming it just one letter, always the same.

tablatronix, 2023-07-04:
This was a htmlentities encoding issue I think. It was fixed if you are seeing this still create a new issue please

commit 32d6f112 (2018-03-21): fixes #573