tzapu/WiFiManager #576: Compile error with core lib 2.3.0

sidey79, 2018-03-26:
## Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.3.0]
- Development Env:		[|other Visual Studio 2017 with Visual Micro]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[1MB]
- lwip Variant:             [unknown]
- Reset Method:		[nodemcu]
- Flash Frequency:	       [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Detailed problem description goes here.

The server->on function in core release 2.3.0 is declared with variables from type char* and not with variables from type string as they are used by this lib.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

example sketches are affected

### Debug Messages

```
WiFiManager.cpp: In member function void WiFiManager::setupConfigPortal()
 
WiFiManager.cpp: 388:78: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(), WiFiManager* const>::type)
   bind(&WiFiManager::handleRoot, this))
WiFiManager.cpp:388: note  candidates are
 
WiFiManager.h:31: In file included from
WiFiManager.cpp:13: from
ESP8266WebServer.h:79: note  void ESP8266WebServer  on(const char*, ESP8266WebServer  THandlerFunction)
   void on(const char* uri, THandlerFunction handler)
ESP8266WebServer.h:79: note    no known conversion for argument 1 from String to const char*
ESP8266WebServer.h:80: note  void ESP8266WebServer  on(const char*, HTTPMethod, ESP8266WebServer  THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn)
ESP8266WebServer.h:80: note    candidate expects 3 arguments, 2 provided
ESP8266WebServer.h:81: note  void ESP8266WebServer  on(const char*, HTTPMethod, ESP8266WebServer  THandlerFunction, ESP8266WebServer  THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn)
ESP8266WebServer.h:81: note    candidate expects 4 arguments, 2 provided```

tablatronix, 2018-03-26:
Thanks looks like this was changed in 2.4.0
https://github.com/esp8266/Arduino/commit/4dc4e75216dfe21ca1bc0ba4eebc2ec94d7b6be5

tablatronix, 2018-03-27:
how in the hell does one install 2.3.0 on platformio.hmm

sidey79, 2018-03-28:
Sorry, no idea.

It's boring, that the upstream lib is changed that way.

tablatronix, 2018-03-28:
I will see about making this backward compatible, not sure how I can use strings and char* and use flash strings

sidey79, 2018-03-28:
The only think i see is, that it is possible to detect the lib version via #ifdef ....

Something like this:
`#ifndef ARDUINO_ESP8266_RELEASE_2_3_0`

And depending on the version use seperate code.

Not very well but then it is backward compatible

abhitheawesomecoder, 2018-08-28:
`Arduino: 1.8.5 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp: In member function 'void WiFiManager::setupConfigPortal()':
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:440:78: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(), WiFiManager* const>::type)'
   server->on((String)FPSTR(R_root), std::bind(&WiFiManager::handleRoot, this));
                                                                              ^
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:440:78: note: candidates are:
In file included from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.h:36:0,
                 from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:441:84: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'
   server->on((String)FPSTR(R_wifi), std::bind(&WiFiManager::handleWifi, this, true));
                                                                                    ^
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:441:84: note: candidates are:
In file included from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.h:36:0,
                 from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);
        ^
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:442:91: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'
   server->on((String)FPSTR(R_wifinoscan), std::bind(&WiFiManager::handleWifi, this, false));
                                                                                           ^
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:442:91: note: candidates are:
In file included from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.h:36:0,
                 from /home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:13:
/home/abhishek/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)
   void on(const char* uri, THandlerFunction handler);
        ^
[... cut ...]
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp: In member function 'bool WiFiManager::WiFi_scanNetworks(bool, bool)':
/home/abhishek/Arduino/libraries/WiFiManager/WiFiManager.cpp:1001:14: error: 'class ESP8266WiFiClass' has no member named 'scanNetworksAsync'
         WiFi.scanNetworksAsync(std::bind(&WiFiManager::WiFi_scanComplete,this,_1));
              ^
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

tablatronix, 2018-08-28:
Yeah the event func was changed in 2.4 and not backward compatible.

tablatronix, 2018-09-07:
using `server->on(String(FPSTR(R_root)).c_str(),   ` for both compatability

bayazidsustami, 2018-10-05:
it's not work on my pc

tablatronix, 2018-10-05:
These changes are only in the development branch at the moment.

MEHUL95, 2018-12-30:
> using `server->on(String(FPSTR(R_root)).c_str(), ` for both compatability

May I know where to add this line in WifiManager.cpp?

tomgillespy, 2019-02-11:
@MEHUL95 if you still need this you see where you need to change it by clicking on the commit hash on the left next to the commit that made the change above. I've just amended the file and it works fine now.

closed 2019-08-30 by tablatronix