tzapu/WiFiManager #577: Custom Parameter Field Description

atatistcheff, 2018-03-27:
I'm not sure if this is a question or a feature request.  The issue I'm running into is that when I add custom fields with the WiFiManagerParameter function the field descriptions are not visible if the field is pre-populated with data.  I've attached a screenshot that illustrates this.  In these custom fields it's easy to see what the empty ones are looking for but in the one that's populated we're really at a loss as to what that is.  To find out we'd have to delete all the contents and then understand that this is the Wunderground API key.  I realize I could copy the existing library and then add my own HTML to the page but I'm wondering if there's any other way to get these field descriptions on the WiFi Manager page other than customizing the library itself?

Thanks for listening!

[image]

atatistcheff, 2018-03-27:
Found this in searching through the previous issues.

Example:
WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "iot.eclipse", 40, " readonly");
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", NULL, 5);
  WiFiManagerParameter custom_text("<p>This is jsut a text paragraph</p>");
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", "sdfsfasdfdsfsdfa", 32);

  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_text);
  wifiManager.addParameter(&custom_blynk_token);

closed 2018-03-27 by atatistcheff

tablatronix, 2018-03-27:
Development version removes placeholders in favor of labels