tzapu/WiFiManager #578: Webconfiguration - HOSTNAME - new feature request

sorriso93, 2018-03-28:
### Basic Infos
NEW FEATURE REQUEST
Hello in addition to webserver ip address & MQTT configuration it would be useful to add hostname, also to be saved as persistent information together with remaining network configuration

tablatronix, 2018-03-28:
You can implement this now in development branch using
custom parameters, spiffs, and WiFiManager.setHostname()

sorriso93, 2018-03-29:
Thanks, I will try but I don't think to be capable of such implementation... Anyway, can you point me to similar addition, if any available on examples or dev branches? 
Thank you in advance

tablatronix, 2018-03-29:
The fsparameters examples are pretty much already there

sorriso93, 2018-04-05:
Many thanks, copied each occurence of mqtt_server and added the same row with new strin variable host_name. Before starting WiFiManager I added:

tablatronix, 2018-04-05:
hostname is included in wifimanager

sorriso93, 2018-04-05:
Using the WiFiManager.setHostname didn't get any change on hostname...

tablatronix, 2018-04-05:
hmm, yeah that makes sense, it only sets hostname for wifimanager, so if calling autoconnect.
are you not using autoconnect?

sorriso93, 2018-04-05:
Yes I used autoconnect and the fsparameter example (I also integrated in my sketch the OTA part in another setup procedure). 
After a lot of trial this is the code doing the job...

void setup_WIFIWEBSERVER ()
{
  Serial.begin(115200);
  Serial.println();
  //clean FS, for testing
  //SPIFFS.format();
  //read configuration from FS json
  #ifdef debug_serial
   Serial.println("mounting FS...");
  #endif
  if (SPIFFS.begin()) {
    #ifdef debug_serial
     Serial.println("mounted file system");
    #endif
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      #ifdef debug_serial
       Serial.println("reading config file");
      #endif
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        #ifdef debug_serial
         Serial.println("opened config file");
        #endif
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);
        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          #ifdef debug_serial
           Serial.println("\nparsed json");
          #endif
          strcpy(host_name, json["host_name"]);
          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(blynk_token, json["blynk_token"]);
          // set hostname
          wifi_station_set_hostname (host_name);
          WiFi.hostname(host_name);        
          if(json["ip"]) {
            #ifdef debug_serial
             Serial.println("setting custom ip from config");
            #endif
            //static_ip = json["ip"];
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
            //strcat(static_ip, json["ip"]);
            //static_gw = json["gateway"];
            //static_sn = json["subnet"];
            #ifdef debug_serial
             Serial.println(static_ip);
            #endif
            //Serial.println("converting ip");
            //IPAddress ip = ipFromCharArray(static_ip);
            //Serial.println(ip);
          } else {
            #ifdef debug_serial
             Serial.println("no custom ip in config");
            #endif
          }
        } else {
          #ifdef debug_serial
           Serial.println("failed to load json config");
[... 97 lines cut ...]
    json["subnet"] = WiFi.subnetMask().toString();
    File configFile = SPIFFS.open("/config.json", "w");
    #ifdef debug_serial
     if (!configFile) 
     {
       Serial.println("failed to open config file for writing");
     }
    #endif
    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }
  #ifdef debug_serial
   Serial.println("\n ----- ");
   Serial.println("hostname");
   Serial.println(WiFi.hostname());
   Serial.println("local ip");
   Serial.println(WiFi.localIP());
   Serial.println(WiFi.gatewayIP());
   Serial.println(WiFi.subnetMask());
   WiFi.printDiag(Serial);
  #endif

}

tablatronix, 2018-04-05:
so you tried

```cpp
wifiManager.setHostname(host_name);
  if (!wifiManager.autoConnect(host_name, password_AP)) 
  {...
```

what did logs say ?

sorriso93, 2018-04-05:
I must admit I don't know how to see log... you mean log of the ESP8266 program execution I imagine...

tablatronix, 2018-04-05:
the serial output

tablatronix, 2018-04-05:
That is the default.

Paste the wm: debug logging if you can test again please, thanks

tablatronix, 2018-06-29:
is this still an issue?

closed 2018-07-02 by sorriso93

DaleSchultz, 2019-04-21:
using Wifimanage 0.14.0 on ESP8266 NodeMCU

```
WiFiManager wifiManager;
wifiManager.setHostname("abcd");
```
results in:

>  error: 'class WiFiManager' has no member named 'setHostname'
>   wifiManager.setHostname("abcd");

tablatronix, 2019-04-21:
Development branch!

DaleSchultz, 2019-04-21:
ah duh!
There seems to be an underlying problem setting hostname() sometimes it works and sometimes not. I suspect it is not a WifiManager issue but numerous people are having the problem. I am giving up.
Thanks.

tablatronix, 2019-04-21:
do you have a reference to problems?

DaleSchultz, 2019-04-21:
there are many threads and issues on it, some of which get confused by looking at what the router sees and reports etc. and many are closed when some workaround is found.

https://github.com/esp8266/Arduino/issues/2826
https://www.esp8266.com/viewtopic.php?f=160&t=15590
https://www.esp8266.com/viewtopic.php?f=29&t=11124
https://www.esp8266.com/viewtopic.php?f=32&t=11306
https://forum.arduino.cc/index.php?topic=558635.0

On Friday this was working fine for me on multiple ESP8266 NodeMCU units:

```
WiFi.hostname("abcd");
Serial.println(WiFi.hostname()); 
```
Then suddenly it no longer works and new downloads produce an empty string. Note if the call to hostname() works, the reading it back with hostname(); should return the name, so there should eb no need to restart routers or load DNS systems etc., those all just confuse and obscure the issue that WiFi.hostname("abcd"); is demonstratively failing on some occasions.

It is erratic, which probably explains why there are so many threads online complaining that it does not work, and it works for others. Some people also report it stopped working after it was working.

In wading through all these thread I came across this thread suggesting to use WiFiManager.setHostname()   as a workaround. It should not be needed.

https://github.com/esp8266/Arduino/issues/2826
has a statement from d-a-v saying

> WiFi.hostname is broken in 2.4.0 with lwip-v2. Please update to latest git or use lwip-v1.4.

but it also fails with LwIP 1.4 for me.

tablatronix, 2019-04-21:
Those threads are very old, dns was broken until like 2.4