tzapu/WiFiManager #579: custom strings doesn't work in VS2017

sidey79, 2018-03-30:
### Basic Infos

Using custom strings doesn't work with VS2017 . I think, with Arduino IDE this is the same.

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266: 

Hardware:			esp-01
Core Version:      	2.4.1 

### Description

Problem description

Adding a  #define WIFI_MANAGER_OVERRIDE_STRINGS in another header file or even in the main .ino file doesn't get recogniced when the WIFIManager lib itself is compiled.

The code example shows, that no error is trown, because the definition is not present inside the wifimanger.h

### Settings in IDE

Module:  			Generic 8266

### Code

```
#define WIFI_MANAGER_OVERRIDE_STRINGS
#include <FS.h>   
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <ArduinoJson.h>     //Local WebServer used to serve the configuration portal

#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
```

tablatronix, 2018-03-30:
What happens if you define in build flags, or define ans include in another header file?

sidey79, 2018-03-31:
Arduino IDE or VS2017 doesn't allow defining build flags easy.
But i am sure, with build flags anything would be fine.

If i put the defines in another header file it is the same.

I think the way to fix this is moving the const vars out of strings_h into wifimanager.h and create #defines for the values.
Defaults can be in anything like en_default_str.h and custom strings should go into user_strings.h or something similar.

It should be possible, to control which one is used via #WIFI_MANAGER_OVERRIDE_STRINGS it the check is done in wifimanager.h.

But i haven't tested that idea.

sidey79, 2018-04-01:
I did some tests yesterday but defines are never populated to libs. There ist no dependency from the lib back to the main Programm.

It is may possible, when moving all the code into the header file. But this isn't good coding style.

Another approach ist to specify a interface object. Instantiate this interface object and pass it as reference in the constructor.
If someone want to change the text, he could define a new class based on the template / Interface.

Going ahead, putting a file on spiffs may also a solution. Putting the file first time on the esp may a little bit tricky.

tablatronix, 2018-07-24:
I have read and tested this over and over and still cannot figure it out. I cannot get a compiler flag in the ini to be recognized in a library.

I have seen it done, the blynk library uses a sketch define for debug mode, and it works.

sidey79, 2018-07-24:
This is, because the header file is compiled independent from the sketch and later linked.

tablatronix, 2018-07-24:
Ok I have it working now.
BUT you cannot define strings in the scope of the library...
ugh

I cannot declare the consts in sketch not in an include file.

```
/Users/.../projects/microcontrollers/dev/libraries/WiFiManager/WiFiManager.cpp: In member function 'String WiFiManager::getHTTPHead(String)':
/Users/..../projects/microcontrollers/dev/libraries/WiFiManager/WiFiManager.cpp:815:17: error: 'HTTP_HEAD_START' was not declared in this scope
page += FPSTR(HTTP_HEAD_START);
```

tablatronix, 2018-07-24:
ugh I dont get C++ linker at all

I thought all consts in `strings_en` would be global and external
```
#include <strings_en.h>
#include <WiFiManager.h>
````

Maybe I need to declare and init seperatly, use flag extern ...
I give up

tablatronix, 2018-07-24:
I am not concerned with defines atm. I am trying to figure oht the scope of the consts in strings.h . 
I thought all includes were precompiled at once as one scope not separately. And  all consts would be global and external.

tablatronix, 2018-07-25:
Ahh I think const are internally linked by default