tzapu/WiFiManager #581: SSID list does not escape HTML

astfgl, 2018-04-02:
### Basic Infos

A SSID containing HTML does not display correctly in list. May make AP name invisible and un-clickable.

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32:		Esp8266

Hardware:			WeMos D1 Mini / ESP-12

Core Version:      	2.4.0

### Description

When a SSID has a name consisting of a HTML-looking tag, the access point may be unuseable. For example a name of **&lt;NULL&gt;** will not be visible in the list. The line for that access point is there, however there is nothing on the line which is clickable, and so the access point is not selectable.

Multiple SSID variants have been tested and all work with other CLI and graphical configuration tools on all platforms tested - Windows, Linux, Android, iOS, etc.

### Settings in IDE

Module:  			Wemos D1 R2 & mini

Additional libraries: NIL

### Sketch

```cpp

#include <Arduino.h>
#include <WiFiManager.h>

void setup() {
  WiFiManager wifiManager;
}

void loop() {

}
```

### Debug Messages

NIL

tablatronix, 2018-04-02:
Thanks I was meaning to add encoding/decoding to the lib and forgot to make an issue for it.

tablatronix, 2018-04-05:
give it a try now

added
```cpp
String WiFiManager::htmlEntities(String str) {
  str.replace("&","&amp;");
  str.replace("<","&lt;");
  str.replace(">","&gt;");
  // str.replace("'","&#39;");
  // str.replace("\"","&quot;");
  // str.replace("/": "&#x2F;");
  // str.replace("`": "&#x60;");
  // str.replace("=": "&#x3D;");
return str;
}
```

astfgl, 2018-04-06:
Works perfectly! Thanks very much for the quick turnaround, much appreciated.

commit c5b36373 (2018-04-05): fixes #581

adds htmlentities to ssid outputs, limited replacements for now
closed 2018-04-06 by astfgl