tzapu/WiFiManager #582: webserver availability once esp8266 joined the wifinetwork

sorriso93, 2018-04-05:
Hello
an example how to use the web server after the WIFiManager configuration to add a simple on/off switch relay?

Just as example, I tried to call server.begin and this code after configuration but the webserver on the ESP8266 doesn't replay to request. Without WifiManager the code is ok and the webserver replay to requests. Where should I put this loop code?

  WiFiClient client = server.available();
  if (!client) {
  return;
  }

  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
  delay(1);
  }

  // Read the first line of the request
  String request = client.readStringUntil('\r');
  Serial.println(request);
  client.flush();

  //Match the request, checking to see what the currect state is
  int value = LOW;
  //if (request.indexOf("/relay=ON") != -1) {
  if (request.indexOf("/on") != -1) {
    digitalWrite(relayPin, HIGH);
  value = HIGH;
  }
  //if (request.indexOf("/relay=OFF") != -1){
  if (request.indexOf("/off") != -1){
    digitalWrite(relayPin, LOW);
  value = LOW;
  }
  // Return the response, build the html page
  //
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");

  client.print("Relay is now: ");

  if(value == HIGH) {
  client.print("Engaged (ON)");
  } else {
  client.print("Disengaged (OFF)");
  }
  client.println("<br><br><br>");
  client.println("<a href=\"/on\">Click here to engage (Turn ON) the relay.</a> <br><br><br>");
  client.println("<a href=\"/off\">Click here to disengage (Turn OFF) the relay.</a><br>");
  client.println("</html>");

  delay(1);
  Serial.println("Client disconnected");
  Serial.println("");

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32:  WEMOS D1

Hardware:			WEMOS D1

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

closed 2018-04-05 by sorriso93

tablatronix, 2018-04-05:
#461 
There has been several discussions on how best to do this, it seems that simply making the webserver public should work

sorriso93, 2018-04-05:
Hi, checked all discussion and tried some code, for example add_endpoint but with no success (I get compilation error 

'class WiFiManager' has no member named 'sendResponse'

in the last row of the callback...

void callback(WiFiManager *wifiManager) {
  //do the work
  digitalWrite(LED, !digitalRead(LED));

  //make the json response
  DynamicJsonBuffer jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json["pin_state"] = digitalRead(LED);
  char buffer[json.measureLength() + 1];
  json.printTo(buffer, sizeof(buffer));

  //send the response back
  wifiManager->sendResponse(200, "application/json", buffer);
}

tablatronix, 2018-04-05:
you would move the webserver pointer to public in the library

and use that, 

`wifiManager->sendResponse` is totally wrong

it should be something like
`wm->WebServer->sendRepsonse`