tzapu/WiFiManager #588: Captive portal not sent after losing connection to AP and power cycle.

Akshay-Rs, 2018-04-10:
if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266: 

Hardware:			ESP-12e

Core Version:      	2.4.1

### Description

I have a NodeMCU that runs WiFiManager before connecting to a REST service. 
These are the steps I follow:
1) Start the captive portal after powering up.
2) Enter the credentials of the AP I want to connect to.
3) The ESP8266 connects to the AP without any problems.
4) I intentionally disconnect the AP. The ESP8266 loses connection.
5) I perform a power cycle on the ESP8266 to restart the captive portal to enter the credentials for another AP. It can't find the AP and starts the captive portal again. However, I am unable to connect to the config portal AP. Even if I manage to connect to the AP, the config page isn't sent to the device.
The serial log, however, has entries stating "Request redirected to captive portal". Unfortunately the portal page was never sent to any of the clients.

I tried the development branch of WiFiManager as well. That couldn't even go past step 3. It gave the status WL_DISCONNECTED for one AP and WL_IDLE_STATUS for another. 
### Settings in IDE

Module:  			NodeMcu v1.0

Additional libraries: ESP8266WiFi, PubSubClient, aRest.h

### Sketch
```cpp
//needed for aRest
#include <ESP8266WiFi.h>         
#include <PubSubClient.h>
#include <aREST.h>

//needed for WiFiManager library
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>         

void setup() {
    //NodeMCU stuff
    Serial.begin(115200);
    pinMode(relay_pin,OUTPUT);
    digitalWrite(relay_pin,HIGH);
    
//arest stuff 
    client.setCallback(callback);
    set_arest_parameters(); 
    char* out_topic = rest.get_topic();
    
    WiFiManager wifiManager;              //WiFiManager
    //reset saved settings
    //wifiManager.resetSettings();
      
    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect();
    
    //if you get here you have connected to the WiFi
        
}

void loop() {
    // put your main code here, to run repeatedly:
    rest.handle(client);
}
```
### Debug Messages

```
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
Attempting MQTT connection...Connected to aREST.io
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
.
(Reset the board)
.
.
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: ESP7688619
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
```

tablatronix, 2018-04-10:
I think I introduced a bug into development, that might be why it was failing, testing now

tablatronix, 2018-04-10:
fixed

Akshay-Rs, 2018-04-10:
Both fixes you suggested worked perfectly. Thanks!

closed 2018-04-10 by Akshay-Rs