tzapu/WiFiManager #590: auto generated accesses point on reset

tusharpc, 2018-04-12:
WiFimanager Branch/Release:  Development

Esp8266/Esp32: ESP8266

Hardware:			ESP-12e

Core Version:      	2.4.1,

### Description

Problem description: 
using development branch and basic auto connect example. Once the device is set to connect to an access point, everything works perfect, reset the NodeMCU and it still connects as client but it simultaneously spins an auto generated access point with default chip name + ID

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}
```

### Debug Messages

```
messages here
```

tusharpc, 2018-04-14:
[image]

tusharpc, 2018-04-14:
[image]

tablatronix, 2018-04-14:
are you setting mode sta in your code? Not sure what esp defaults to after you erase flash

tablatronix, 2018-04-14:
ill look into any bugs , i had a way to fix this but it is disabled because it was causing issues, so it might be wm.

tusharpc, 2018-04-14:
@tablatronix , not sure what you mean "setting mode sta in your code?", pretty much using the standard example, can you point me to some examples? (I'm a bit slow on the coding side ) .
btw this issue is not happening on the master version, but it has the issue of not being able to reach the config portal if the original AP is unavailable, you described a solution here : #525

also, just unbelievably awesome work you (and Tzapu ) doing here Shawn. I think it has truly opened up the esps to be IOT's for the masses.

tablatronix, 2018-04-14:
The master version changes your modes , the development version should not, so you are probably seeing the correct behavior now. 

In your code call WiFi.mode(WIFI_STA) and it will remember it even if you remove it after running it.

I will try to fix the feature that lets you set the preffered mode after wm exits. Its disabled it crashes stuff. Also will check for any bugs that might be changing modes

tusharpc, 2018-04-14:
@tablatronix thanks i will try that out, post on results shortly.

closed 2018-04-16 by tusharpc

iamjameshannam, 2018-05-28:
Is there  a way to grab the device ID and create a unique device name on the splash screen? For example, MyDevice_1234?