tzapu/WiFiManager #591: How to disconnect wifi and restart http server

kotran88, 2018-04-13:
----------------------------- Remove above -----------------------------

node mcu

and I'm using it with autoConnect("myWifi") 
then connect wifi very well. 

but I have disconnect button to disconnect. 
if I press disconnect button, 

wifiManager.resetSetting();
ESP.reset()

and Serial monitor says below. 

*WM: settings invalidated

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

That's all! 

I think It should restart http server to connect it right away. 
but it stop wdt reset log. 
I should build code again to connect wifi again. 

How can I reset wifi and connect it again ?

tablatronix, 2018-04-13:
Does it not autoconnect? Are you on usb and programming? You have to reset the module after programming for resets to work, its a esp known issue

kotran88, 2018-04-14:
for wifiManager.resetSetting();
it 's not goes to  auto connect automatically. but only shown *WM: settings invalidated. 
reset the module mean ESP.reset()???

tablatronix, 2018-04-14:
If you are programming an esp over serial, you cannot soft reset until you hard reset. It is a limitation.
Hit reset after or repower it then try

kotran88, 2018-04-14:
I added question...can you check image and review it...? it's same reason? for soft reset, I should hard reset.

tablatronix, 2018-04-14:
YES

n4rf, 2019-06-04:
Sorry, Im quite late. But I have been sketching on a mesh of Wifi sensors that as far as it goes it works with no problems! Wifimanager did the trick quite flawlessly.
 The only counterback I am having is that when ESP32 can not connect (usually becasue I am testing it for possible reasons it may not connect, such as password has changed), it enters on AP mode to configure it. So far so good.
When no Wifi settings (SSID/PASSWD) has been saved, it works flawlessly.
But, if there has been a previous (now not working) Wifi setting, when AP mode starts, and I connect to it, I have hard time to access the configuration page. Usually it hangs, or disconnects. Therefore I thought about making a "Reset configs" instance, to erase all previous settings.

If i manually erase the ESP32 flash, everything works well. 
But, I dont want to have to manually erase each ESP32 everytime Wifi config changes.

Carefully reading the library, I came across this:

void WiFiManager::resetSettings() {
  DEBUG_WM(F("settings invalidated"));
  DEBUG_WM(
      F("THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT "
        "OUT AFTER ERASING THE DATA."));
  // TODO On ESP32 this does not erase the SSID and password. See
  // https://github.com/espressif/arduino-esp32/issues/400
  // For now, use "make erase_flash".
  WiFi.disconnect(true);
  // delay(200);
}

Is there any chance this has had any progress? Or any suggestions as how I may aproach it? 

Sorry for the issue, thanks in advance for the library and al the work done so far.

n4rf, 2019-06-06:
After reverse-engineering the ESP32 library I found a "quick-n-dirty" way to erase the saved WiFi Credentials that messed up with the access to the configuration portal.

Initially, WiFi.disconnect() does not work unless, your are connected, of course, but I needed a way to erase credentials because I am using Wifimanager and since I was not connected, and needed to save a new one, old credentials messed up the AP and STA mode, hence, I needed to erase previous saved credentials.

the trick is quite simple and it was in front of my eyes the whole time.
```

        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); //load the flash-saved configs
        esp_wifi_init(&cfg); //initiate and allocate wifi resources (does not matter if connection fails)
        delay(2000); //wait a bit
        if(esp_wifi_restore()!=ESP_OK)
        {
            Serial.println("WiFi is not initialized by esp_wifi_init ");
         }else{
             Serial.println("WiFi Configurations Cleared!");
         }
         //continue
        delay(1000);
        esp_restart(); //just my reset configs routine...
```

And that will erase the previously saved WiFi credentials and let me acccess to the configuration portal without problems

tablatronix, 2019-06-06:
What is wrong with wm.resetSettings()

n4rf, 2019-06-06:
My setup to try if it worked (after my assumption it did not) was to setup a working SSID and Passphrase, and then wm.resetSettings(), it was supposed to erase the config, and therefore connection would have failed after restart. I kept reading on my serial monitor that connection status was "Connected" and pinging my server returned OK, so, config was not dropped.

On the library I am using, on WiFiManager.cpp on line 357 I found:

```
void WiFiManager::resetSettings() {
  DEBUG_WM(F("settings invalidated"));
  DEBUG_WM(
      F("THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT "
        "OUT AFTER ERASING THE DATA."));
  // TODO On ESP32 this does not erase the SSID and password. See
  // https://github.com/espressif/arduino-esp32/issues/400
  // For now, use "make erase_flash".
  WiFi.disconnect(true);
  // delay(200);
}
```
I read on other post that WiFi.disconnect() needs to be connected in order to work properly, which in my case, there is no guarantee SSID and passphrases would be permanent. 

Like I said earlier, I needed to erase the config without a successful connection, reading the library it seems there is some sort of connect check before WiFi.disconnect cant do its job. 

On WiFiSTA.cpp (esp32 WiFi library), on line 237 I found:

```
bool WiFiSTAClass::disconnect(bool wifioff, bool eraseap)
{
    wifi_config_t conf;
    if(WiFi.getMode() & WIFI_MODE_STA){
        if(eraseap){
            memset(&conf, 0, sizeof(wifi_config_t));
            if(esp_wifi_set_config(WIFI_IF_STA, &conf)){
                log_e("clear config failed!");
            }
        }
        if(esp_wifi_disconnect()){
            log_e("disconnect failed!");
            return false;
        }
        if(wifioff) {
             return WiFi.enableSTA(false);
        }
        return true;
    }

    return false;
}
```

Of course, the eraseap is supposed to force the AP erase..   I tried WiFi.disconnect(false,true).. (true,true), (), etc.. same result every time... it kept connecting to my network after reset even after the instruction to erase the AP.

So, going to the basics, I found the esp_wifi_restore(), which needed esp_wifi_init() to allocate memory and the resources.

> esp_err_t esp_wifi_restore(void)
> 
>     Restore WiFi stack persistent settings to default values.
> 
>     This function will reset settings made using the following APIs:
> 
>         esp_wifi_get_auto_connect,
>         esp_wifi_set_protocol,
>         esp_wifi_set_config related
>         esp_wifi_set_mode
> 
>     Return
> 
>             ESP_OK: succeed
>             ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
> 
> 

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html
 
I found that WiFi.begin() uses this:

```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); 
esp_wifi_init(&cfg);
```
So after those instructions and a little delay it worked!

Final routine code to "restore to default WiFi settings"

during setup:

```
    if(!digitalRead(btnOK.PIN)){  //if its pressed while turning on, its for reset purpouses!
    unsigned long reset_t=0;
    while(!digitalRead(btnOK.PIN)){
      delay(10);
      reset_t++;
      if(reset_t>200){ //more than 2secs hold
        display.clearDisplay();
        display.setTextSize(2);             // Normal 1:1 pixel scale
        display.setTextColor(WHITE);        // Draw white text
        display.setCursor(30,30);             // Start at top-left corner
        display.print("Reset!");
        display.display();
        delay(200);
        for(int i=0;i<=1023;i++){  //erase al eeprom saved data
          EEPROM.write(i,0x255);
          EEPROM.commit();
        }
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); 
        
        esp_wifi_init(&cfg);  //load resources and allocate memory
        delay(2000);
        if(esp_wifi_restore()!=ESP_OK)
        {
          Serial.println("WiFi is not initialized by esp_wifi_init ");
          }else{
          Serial.println("WiFi Configurations Cleared!");
            }
        delay(1000);
        esp_restart();
      }
     }   
    }
```

tablatronix, 2019-06-07:
No idea what library you are using.... its not this one..

```C++
/**
 * [resetSettings description]
 * ERASES STA CREDENTIALS
 * @access public
 */
void WiFiManager::resetSettings() {
  DEBUG_WM(F("SETTINGS ERASED"));
  WiFi_enableSTA(true,true); // must be sta to disconnect erase
  
  if (_resetcallback != NULL)
      _resetcallback();
  
  #ifdef ESP32
    WiFi.disconnect(true,true);
  #else
    WiFi.persistent(true);
    WiFi.disconnect(true);
    WiFi.persistent(false);
  #endif  
}
```

pmartinz, 2020-10-28:
Sorry that this is of topic but I'm not sure how to contact with private message. @n4rf I'm trying to do what you have done, ESP32, WiFiManager and mesh (painlessmesh) but I'm not able to get it to work. Is it possible for you to share this part of your code? Like how you do the sequence in the start. Init -> WiFiManager -> mesh