tzapu/WiFiManager #593: Arduino_esp8266

atomatick12, 2018-04-17:
when i used DoubleResetDetector.h library in code i got following error

error: 'class EspClass' has no member named 'rtcUserMemoryRead' ESP.rtcUserMemoryRead(address, &doubleResetDetectorFlag, sizeof(doubleResetDetectorFlag)

tablatronix, 2018-04-17:
Wtf is double reset detector?

liebman, 2018-04-17:
@atomatick12 I think what @tablatronix is asking is what does DoubleResetDetector have to do with WiFiManager?

atomatick12, 2018-04-18:
i am using it for node mcu esp8266 builtin reset button for wifimanager re configuration

atomatick12, 2018-04-18:
i just want to use builtin reset button to one time press for reset and two time press to AP mode for configration

tablatronix, 2018-04-18:
Ah I see, that library uses rtc I wrote a sketch that uses eeprom and takes several resets.

Ill take a look at it

atomatick12, 2018-04-19:
i just want to use builtin reset button to one time press for reset and two time press to AP mode for configration any body help on this

tablatronix, 2018-04-19:
Were you able to get the example to compile?

tablatronix, 2018-08-16:
did you resolve this?