tzapu/WiFiManager #595: OnDemandWebPortal example missing information on what happens when NOT in config mode?

schweini, 2018-04-19:
Maybe I'm being denser than usual, but neither the examples for the on-demand config portals or the documentation as such explain what happens if the user is NOT in config mode.

I need to try to connect to the last successful network. If this doesn't work, then I only want to start the config mode when certain criteria are met (e.g. button press).

How am I supposed to connect to the last known working wifi network, and only start the config mode on failure?

tablatronix, 2018-04-20:
besides using the standard code wifi.status() == wl_connected loop ?

schweini, 2018-04-20:
Thanks for your quick reply!

Ideally, I'd like to use WifiMulti (seems to be the standard way of manually connecting nowadays?) to connect to some Wifi networks. Only if there is none of the predefined networks available, I'd like to use WifiManger to start the Config Portal, and configure the connection details using the web interface.

What I don't understand is how I can get the ESP32 to 'know' when it has been configured using WifiManager, and use this information instead of WifiMulti afterwards?

In pseudo-code, I guess what I'm looking for is how to do this:

`if ( wifiManger.connectionSuccessful() ) 
{
  connected = true;
}
elsif ( WifiMulti.connectionSuccessful() )
{
  connected = true
}
else
{
  if (button_pressed)
  {
  wifiManager.startConfigPortal("OnDemandAP")
  }
}
`

tablatronix, 2018-04-20:
Wifi multi is for connecting to mutiple aps, and it only stores the last successful one to flash, so its not really the greatest thing,unless you are actually connecting to mutiple aps.

The connection status is the return value of autoconnect. 

There is no way to differentiate what saved the ssid in memory though.

schweini, 2018-04-20:
But even I were not to use WifiMulti and just regular Wifi.begin(), how can I tell wifiManger to attempt to connect using its stored credentials, and if unsuccessful, return control to my program, so that I can decide when to call startConfigPortal()?

As far as I can see, I only have the option of calling autoConnect() (which attempts the connection, and starts the Config Portal if unsuccessful), or to not connect at all, and start the Config Portal manually.

Because of you mention of ....

Oooops. Just re-read your other comment, and the mention of Wifi.begin() without arguments. Will try that!

tablatronix, 2018-04-20:
Are you using development branch?

tablatronix, 2018-04-20:
If so
‘setEnableConfigPortal’

closed 2018-06-29 by tablatronix