tzapu/WiFiManager #597: SPIFFSReadServer support for page template.

MiggyMan, 2018-04-21:
It would be nice to have support for the SPIFFSReadServer library (or similar functionality)  to store the html for the wifi manager in spiffs to make it a little easier to work with!

tablatronix, 2018-04-22:
But wm doesnt serve static files, I mean css maybe but whats the point of bloating your lib for that

MiggyMan, 2018-04-22:
Customisation, and the point of using spiffs is to avoid bloat and give more scope for customisation, whacking html in variables is not optimal, or in anyway fun to work on, loading a file from a local filesystem makes it far easier to adapt the page to fit your project as well removing the overhead to have the html stored in program space.

tablatronix, 2018-04-22:
I still dont get it, its still gonna be html fragments and still going to need a parser and a templating mechanism

MiggyMan, 2018-04-22:
I'll look into using PersWiFiManager, I much prefer this but hey-ho.

closed 2018-04-22 by MiggyMan

tablatronix, 2018-04-22:
If you have a more concise proposal about how exactly this is supposed to work, then provide it. Have you checked the development branch?