tzapu/WiFiManager #598: Slower than regular connection.

matphillips, 2018-04-24:
Simple code included to demonstrate.
Connecting to the wifi on power up (or wake from deep sleep) is slower when using wifimanager.

The demo sketch takes about 5 second to print the IP address and go to sleep. Press the reset button, watch it do it again in five seconds. Change the define so it does not use wifimanager, and the process takes two seconds.

Arduino 1.8.5
WifiManager 0.12.0
Core 2.4.1
ESP8266's : Wemos D1 and NodeMCU

sample code: (simplified to the extreme, may need a fully featured sketch running first to set the wifimanager settings)
```
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#define useWifiManager 1

const char* wifiSsid = "Unit-4x";
const char* wifiPass = "oakadmin";

void loop() {}
void setup() {
  Serial.begin(115200);
  Serial.print("Connecting to ");
  Serial.println(wifiSsid);
  
  #if useWifiManager == 0
    // Normal wifi
    WiFi.mode(WIFI_STA);
    WiFi.begin(wifiSsid, wifiPass);
    while (WiFi.status() != WL_CONNECTED) {
      delay(100);
      Serial.print(".");
    }
  #else
    // WIFI MANAGER
    WiFiManager wifiManager;
    wifiManager.autoConnect();
  #endif

  Serial.println(WiFi.localIP());
  ESP.deepSleep(0, WAKE_NO_RFCAL);
}
```

tzapu, 2018-04-24:
why not start wifimanager only if this fails then?
```
while (WiFi.status() != WL_CONNECTED) {
      delay(100);
      Serial.print(".");
    }
```

matphillips, 2018-04-24:
Interesting, I had no idea that would be possible, I thought the said and password would have to be in the code for the basic wifi to work. I'm testing this out, looks like a good solution.

matphillips, 2018-04-24:
Yup. Looks good, the following barebones code will connect to wifi if it know about it from the wifimanager, and if it knows nothing or cannot connect after 10 seconds, it will launch wifi manager. 

This will work well for my purposes, and may help others.
It's a work around though, but possibly a completely valid one.

```
#include <ESP8266WiFi.h>
#include <WiFiManager.h>

void loop() {}
void setup() {
  Serial.begin(115200);
  Serial.print("Connecting to ");

  WiFi.mode(WIFI_STA);
  unsigned long wifiConnectStart = millis();
  while (WiFi.status() != WL_CONNECTED) {
    if (WiFi.status() == WL_CONNECT_FAILED) {
      return;
    }
    if (millis() - wifiConnectStart > 10000) {
      WiFiManager wifiManager;
      wifiManager.autoConnect();
    }
    delay(100);
    Serial.print(".");
  }
  
  Serial.println(WiFi.localIP());
  ESP.deepSleep(0, WAKE_NO_RFCAL);
}
```

tablatronix, 2018-04-24:
Try development branch?

You also might want to just make sure to add a delay to give wifi a chance to autoconnect before calling wifimanager, since it should not connect at all if already connected.

matphillips, 2018-04-24:
Thanks, yes, will test the timings with the development branch and see what happens. I tried rolling the changes into my main code and messed it up (!) so will re-check this in the morning and update&close the issue with the results, if that's ok?

tablatronix, 2018-04-25:
Did you disable debugging, serial print is slow

tablatronix, 2018-04-25:
I have figured out what the delay is.

`  WiFi_Disconnect(); // disconnect before begin, in case anything is hung`

we disconnect and reconnect, also I was right wifi is not auto started before autoconnect then we try to connect also. But that is not much of a problem, I get 3-5 seconds with that line , commented out I get 1500-3000 ms

tablatronix, 2018-04-25:
we can do some optimizations here also
* If time from boot is n, wait for connectresult change before continue
* If esp is connecting ( no idea how to detect this yet ) then as above.
* detect what reasons we need to disconnect for, and avoid it when possible.

tablatronix, 2018-04-25:
also f you want to shave 100 ms do not use autogenetated ao names, set one in autoconenct()

matphillips, 2018-04-25:
Nice one!
I'll set an AP name, every little helps ;)

tablatronix, 2018-04-25:
I routinely get the same speed as no wifimanasger 1005 ms, sometimes 3000 but that seems to be a fluke

tablatronix, 2018-04-25:
It looks like wifi begin sets temp status WL_IDLE_STATUS until either connect or retry fail, i do not know if autoconnect does this however, since it is sdk initiated, I will have to test, it will probably be safer to avoid disconnect, for all i know this workaround is no longer needed woth current versions.

tablatronix, 2018-04-25:
Commit ref
https://github.com/tzapu/WiFiManager/commit/179c0309a5c1a894bdc8707d901ffb6677888bbf
I did not say why I added that

CrispinP, 2020-01-23:
> I routinely get the same speed as no wifimanasger 1005 ms, sometimes 3000 but that seems to be a fluke, so even with debugging on, that disconnect is the biggest issue.
> 
> i will keep this issue open and attempt some resolutions , we can at least make that disconnect optional if we have to .

@tablatronix,

What WiFi kit are you using? Is it Unifi by any chance? I've the exact timing issue. Mostly at 1000ms-ish but sometimes up to 10 seconds. On the bench, when testing, 230ms. 

Being battery operated, this is a killer :(

tablatronix, 2020-01-24:
Ill have to reread the issue but there are several things of note,  we disconnect before connecting, this can be disabled and is slow, but more stable.

Some changes to handle the async properties of set mode in the esp lib adds a timer to wait here so it can cause some variability in connect times.

For battery I suggest not using autoconnect! Use your own logic and let esp auto connect and use button actions to launch cp or use a timer before starting it.  Also for power saving always use static ip, dhcp is slow