tzapu/WiFiManager #599: first http request is slow than following ones

sohaibilyas, 2018-04-25:
i edited the library files just to add few more buttons with javascript ajax alright so when i open the config portal with ip it takes around 5-10 seconds for the first time so its like when i turn off wifi on my phone come back again in few minutes, connect to it and then again it takes the same amount of time but just with the FIRST REQUEST so does esp8266 sleep when no requests are coming for specific period of time or what! kindly guide me on this and thank you so much for this amazing library <3

sohaibilyas, 2018-04-25:
master branch, haven't tested other branches like dev or hotfixes? ive seen something like scanning aps in [asyncwifimanager](https://github.com/alanswx/ESPAsyncWiFiManager/blob/master/ESPAsyncWiFiManager.cpp) (line 443-449) maybe developer added that to make the esp never sleep! could we also do that in this library!

tablatronix, 2018-04-25:
You can try disabling sleep mode, I think the default is `modem`

`wifi_set_sleep_type(NONE_SLEEP_T);`

tablatronix, 2018-04-25:
This issue seems to be maybe related?
https://github.com/esp8266/Arduino/issues/2330

sohaibilyas, 2018-04-25:
alright im going to try this now, i could try dev branch without any documentations but not really familiar with c++, thanks :)

tablatronix, 2018-04-25:
Yeah no prob, it should work the same, it just some new stuff added
You should be able to do that in your own code to set the sleep in setup.

#500

sohaibilyas, 2018-04-30:
this is just amazing, it was my phone, huawei p10 lite, i tried with my brother's huawei mate 10 lite and i got the portal right away so i think my phone has more apps than my brother's and those apps may be sending too many requests in the background as soon as it gets connected to any wifi :) for now closing it because it is still "may be" 😄

closed 2018-04-30 by sohaibilyas