tzapu/WiFiManager #600: setConfigPortalTimeout() does not reattempt connection; causes strange behaviour

cjastone, 2018-04-25:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			ESP-09

Core Version:      	2.4.0, staging

### Description

When using setConfigPortalTimeout(), following timeout, connection is not reattempted, nor is the device rebooted when the timeout is reached.

The program execution continues, but behaves... strangely.  Most of my code runs within a `while (WiFi.status() == WL_CONNECTED)` loop with not much else happening outside of this within `void loop()`.  So it will sit there and loop a few times, not doing much.  Sometimes, it'll crash after several loops - but other times it suddenly gets a connection to WiFi, but nothing works right - JSON objects are full or gibberish, other variables don't contain correct data, and so on.

I've got a fair bit of code that would be excessive to post here, but if it'll help I can try trim it down to a proof-of-concept.

tablatronix, 2018-04-25:
Not sure what you expect to happen here but timeout just returns the function call, its times out the configportal and shuts it down.

What are you expecting to happen, I do not understand.

Sounds like something is wrong with your code or scope.

cjastone, 2018-04-26:
Okay - understood.  I misinterpreted what was meant by "When the time passes, the autoConnect function will return, no matter the outcome. Check for connection and if it's still not established do whatever is needed (on some modules I restart them to retry, on others I enter deep sleep)"

I've since added in code to handle this myself `if (WiFi.status() != WL_CONNECTED) { ESP.restart(); }` which deals with it just fine.  Thanks for clarifying!

closed 2018-04-26 by cjastone

tablatronix, 2018-04-26:
Yeah that is exactly what the example does