tzapu/WiFiManager #601: Obtain WiFi channel after connecting to AP?

debsahu, 2018-04-25:
After successful connection to AP using WiFiManager, is there an easy way to get the channel of the AP it is connected to? `WiFi.SSID()` and `WiFi.psk()` give SSID and password back but can't get channel number. Since WiFiManager does a scan, can I get the channel number of the connected network using ESP8266WiFiSTAClass?
```
int n = WiFi.scanNetworks(false, true); // This step is already done in WiFiManger?
String ssid; uint8_t encryptionType; int32_t RSSI; uint8_t* BSSID; int32_t channel; bool isHidden;
for (int i = 0; i < n; i++) {
  WiFi.getNetworkInfo(i, ssid, encryptionType, RSSI, BSSID, channel, isHidden);
  Serial.printf("%d: %s, Ch:%d (%ddBm) %s %s\n", i + 1, ssid.c_str(), channel, RSSI, encryptionType == ENC_TYPE_NONE ? "open" : "", isHidden ? "hidden" : "");
}
```

tablatronix, 2018-04-25:
` Serial.println(WiFi.channel());`

debsahu, 2018-04-25:
I thought channel class needed a `networkItem`, according to https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/scan-class.html#channel I might be wrong.

tablatronix, 2018-04-25:
```cpp
/**
 * Return the current channel associated with the network
 * @return channel (1-13)
 */
int32_t ESP8266WiFiGenericClass::channel(void) {
    return wifi_get_channel();
}
```

debsahu, 2018-04-25:
you the best, thank you. close?

closed 2018-04-25 by debsahu

tablatronix, 2018-04-25:
now the scan class does have its own channel function also which is what you posted above