tzapu/WiFiManager #603: INcluding max6675 library invalidates settings

PiteousHonking, 2018-04-30:
----------------------------- Remove above -----------------------------

### Basic Infos

Adding the max6675 library to a sketch causes settings invalidated error

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			Esp8266 WemosD1
https://github.com/adafruit/MAX6675-library 1.0 installed
Arduino 1.8.5 IDE

### Description

The attached code works as expected, creates an AP, stores wifi settings and Blynk auth token and connects to Blynk server sucessfully

If I include the max6675.h library ait will run through the same process but fail to connect to the Blynk server with the error:

*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
(⸮)⸮R(⸮⸮⸮

Comment out the max665.h library  and it works again. Also posted on the blynk forum but it seems to be a wifimanager/max6675 issue. 

### Settings in IDE

Module:  			Wemos D1

Additional libraries: https://github.com/adafruit/MAX6675-library

### Sketch

```cpp

char blynk_token[33];
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <BlynkSimpleEsp8266.h>

// Needed for  the Thermocouple
#include <max6675.h>
int thermoDO = D5;
int thermoCS = D6;
int thermoCLK = D7;
MAX6675 thermocouple(thermoCLK, thermoCS, thermoDO);
int vccPin = 3;
int gndPin = 2;

void setup() {
    Serial.begin(115200);  
    //pinMode(vccPin, OUTPUT); digitalWrite(vccPin, HIGH); //Thermo
    //pinMode(gndPin, OUTPUT); digitalWrite(gndPin, LOW);  //Thermo
    WiFiManagerParameter custom_blynk_token("Blynk", "blynk token", blynk_token, 33);
    WiFiManager wifi;
    wifi.addParameter(&custom_blynk_token);
    wifi.autoConnect("Blynk");
    Blynk.config(custom_blynk_token.getValue());
}
void loop()
 {
  Blynk.run();
}
```

### Debug Messages

```
mounting FS...
mounted file system
reading config file
opened config file
{"blynk_token":"REMOVED"}
parsed json
*WM: Adding parameter
*WM: blynk
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.53
connected...yeey :)
[9633] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.5.2 on Arduino

[9745] Connecting to blynk-cloud.com:80
BLYNK Connection Fail
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.

```

felipepereira92, 2018-06-17:
I'm also with this problem. We need help!!!

PiteousHonking, 2018-06-17:
Can you expand on what you mean, is there a way of determining what it is that's calling resetSettings?

tablatronix, 2018-06-17:
That sketch is surely not your entire code

tablatronix, 2018-06-17:
I cant even get blynk to connect, it just sits in blynk.connect loop.
hmm

tablatronix, 2018-06-17:
ahh I forgot #define BLYNK_PRINT Serial

tablatronix, 2018-06-17:
I cannot reproduce so this must be in your code. Not sure what else to test here.

felipepereira92, 2018-06-17:
I got, i include the send sensor, BlynkTimer and i cheked the pins (D4, D5 and D6) because i had chnged.

`//#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <EEPROM.h>
#include <max6675.h>

BlynkTimer timer;

int thermoDO = D4;
int thermoCS = D5;
int thermoCLK = D6;
MAX6675 thermocouple(thermoCLK, thermoCS, thermoDO);

void sendSensor();
// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "TOKEN";

void setup()
{
  EEPROM.begin(512);
  Serial.begin(115200);
  WiFiManager wifiManager;
  //wifiManager.resetSettings();    //Uncomment this to wipe WiFi settings from EEPROM on boot.  Comment out and recompile/upload after 1 boot cycle.
  wifiManager.autoConnect("NodeMCU");
  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  //pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output, I like blinkies.
  Blynk.begin(auth, WiFi.SSID().c_str(), WiFi.psk().c_str());
  timer.setInterval(1000L, sendSensor);
}

void loop()
{
  timer.run();
  Blynk.run();  // Commented out for a short time.
}

void sendSensor(){
 Blynk.virtualWrite(V5, thermocouple.readCelsius());
}`

consolacion, 2018-10-06:
I don't think it is 'his code'  I have the same problem with the unaltered example code "autoconnect with custom parameters"

tablatronix, 2018-10-06:
Settings invalidated come from wm.resetSettings
There is no other reason for it occur unless you are calling it

consolacion, 2018-10-06:
Thanks, I see the example called that 'for testing'.
I understand now. Learning every day :-)

tablatronix, 2018-10-06:
I just commented it out in that one example in developement , it was not supposed to be active

consolacion, 2018-10-06:
That might be a good idea. I overlooked it as I thought that was what the
SPIFFS.format();
was for :-) But I start to understand the code better now. Tnx