tzapu/WiFiManager #614: WiFi Connection Confirmation

botletics, 2018-06-01:
Hi there,

It would be cool to add the option of having a confirmation message when the module actually successfully connects to WiFi after the settings are sent by the user in the WiFi settings. Currently it only shows the page that says "Credentials Saved!" but if it fails, there's no way the user knows unless they're reading the debug logs in serial monitor or looking at an LED. So that's where the confirmation would come in handy.

What's the best way to do this?
Thanks!

tablatronix, 2018-06-02:
I have been thinking about this, its a bit hard since ap disconnects to sta can connect.

botletics, 2018-06-02:
Yea I'm aware of that as well, it starts in AP but then connects to the WiFi network. Wonder if there's a good way of doing that.

tablatronix, 2018-06-02:
I started adding ajax and status checking to development branch to start to play with grabbing it that way , and having the page js retry until it gets an answer, we can always assume that it was successful if the webpage never comes back..

Or we can restart the ap if success, this has to be tested since mode_ap_sta can cause some channel issues and be problematic sometimes.

tablatronix, 2018-06-02:
I think keytaylor tried this in his branch , but I have not looked at it in awhile. Not sure of the success

hochmayrfa, 2018-07-04:
Any update on this feature ?

tablatronix, 2018-07-04:
Not yet, I am preparing release, working on examples and docs and my youtube channel.
I will try to test out my ideas soon

tablatronix, 2019-12-04:
#930

tablatronix, 2019-12-04:
I have explained this conundrum in another issue, when I find it I will repost it here

garageeks, 2019-12-11:
I have been using kentaylor fork for some time, and works seamlessly; it is more practical from a user perspective.
However it hasn't been updated for 3 years, it would be awesome if this feature could be backported. Let me know how I can be helpful with this.

Nick

tablatronix, 2019-12-12:
How does it work? Does he just have an ajax refresh wait, and have the server cycle back on after connecting to sta ?