tzapu/WiFiManager #617: Reload SSID & psk when configuring

dontsovcmc, 2018-06-07:
----------------------------- Remove above -----------------------------

### Basic Infos
I don't see SSID and password when I go to configure device second time. 

#### Hardware
WiFimanager Branch/Release:  
master 

Hardware:		NodeMCU 2 ESP-1E

Core Version:      	2.4.0

ide:                          Platformio

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu v2

tablatronix, 2018-06-08:
They are not overwritten when empty so whats the problem

dontsovcmc, 2018-06-08:
I need to enter ssid & psk second time.

tablatronix, 2018-06-08:
What do you mean you need to enter second time ?

dontsovcmc, 2018-06-08:
When I turn on AP and open 192.168.4.1 second time I see custom parameter values. The SSID and passwork EditBoxes are empty.

tablatronix, 2018-06-08:
Yeah they are supposed to be. Still do not see the problem here. They are still set just are not shown

dontsovcmc, 2018-06-08:
"They are still set just are not shown". Oh, I don't know this fact. It's good.
Maybe it's better to show them?

tablatronix, 2018-06-09:
Maybe try the development version. Password is hidden on purpose still however due to lack of ssl

dontsovcmc, 2018-06-09:
There is no in release 0.12. Waiting 0.13...

closed 2018-06-09 by dontsovcmc

tablatronix, 2018-06-09:
Why wait, use development branch