tzapu/WiFiManager #618: Unable to access Static IP assigned to ESP8266

Paraagsj, 2018-06-09:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266 

Hardware:			ESP-12e

Core Version:      	2.4.0, staging

### Description

I am trying to assign a static IP to my ESP. For this I have been using the function wifiManager.setSTAStaticIPConfig(IPAddress(192,168,0,122),IPAddress(192,168,0,1),IPAddress(255,255,255,0));
But after getting connected to the Wifi via APconnect the said IP i.e. 192.168.0.122 is either unreachable/Unable to ping/pinging but unable to respond.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#define LED1 D4
#define LED2 D2
 
//const char* ssid = "TP-LINK_4376";
//const char* password = "37244873";
 
// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);
 
void setup() {
  Serial.begin(115200);
  delay(10);
 
  // prepare GPIO2
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  digitalWrite(LED1, 0);
  digitalWrite(LED2, 0);
  
  // Connect to WiFi network
  //Serial.println();
  //Serial.println();
  //Serial.print("Connecting to ");
  //Serial.println(ssid);
  
  //WiFi.begin(ssid, password);

  WiFiManager wifiManager;
  wifiManager.setSTAStaticIPConfig(IPAddress(192,168,0,122),IPAddress(192,168,0,1),IPAddress(255,255,255,0));
  wifiManager.autoConnect("AutoConnectAP");
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  // Start the server
  server.begin();
  Serial.println("Server started");
 
  // Print the IP address
  Serial.println(WiFi.localIP());
}
 
void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
  // Wait until the client sends some data
  Serial.println("new client");
  while(!client.available()){
    delay(1);
  }
  
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  
  // Match the request
  int val1;
  int val2;
  if (req.indexOf("/led1/0") != -1)
    val1 = 0;
  else if (req.indexOf("/led1/1") != -1)
    val1 = 1;
  else if (req.indexOf("/led2/0") != -1)
    val2 = 0;
  else if (req.indexOf("/led2/1") != -1)
    val2 = 1;
  else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

   
  // Set GPIO2 according to the request
  if (req.indexOf("/led1/") != -1)
  digitalWrite(LED1, val1);
  else
  digitalWrite(LED2, val2);
  
  client.flush();
 
  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val1)?"high":"low";
  s += (val2)?"high":"low";
  s += "</html>\n";
 
  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");
 
  // The client will actually be disconnected 
  // when the function returns and 'client' object is detroyed
}

### Debug Messages

Server started on 192.168.0.122

tablatronix, 2018-06-10:
What does your serial logs say? Try development branch

Paraagsj, 2018-06-13:
@tablatronix 
Sharing the logs of my serial monitor below.
As you can see i was able to get connected and it shows that IP 10.0.1.78 has been assigned to the ESP, but when i try accessing that the concern web page does not even respond.
Is there some problem with my code?
This time I have followed the with development branch of WifiManager.
You can check my code in the post which I have updated.

Serial monitor logs:
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with MotoG3 7891, channel 9
ip:10.0.1.78,mask:255.255.255.0,gw:10.0.1.1
ip:10.0.1.78,mask:255.255.255.0,gw:10.0.1.1
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 10.0.1.78
connected...yeey :)
local ip
10.0.1.78
pm open,type:2 0

tablatronix, 2018-06-13:
Does it work with dhcp?

Paraagsj, 2018-06-15:
Yeah it works with DHCP by allocating a random IP to esp. But that's not what I want. I want a specific IP to be assigned to ESP which will be programmed already.

RmenaLopez, 2019-07-04:
No new comments on this? I have the exact same issue

tablatronix, 2019-07-05:
Can you try hotfixes branch ( there was something that was fixed I think with ip being applied on reconnects or something )
also try development branch

pvihang, 2019-07-31:
Dear All
I am new to this community, but have been working on arduino and nodemcu programming for a while.
Currently in one of my project on nodemcu, I need to have a static ip assigned using WiFiManager
The dhcp version of the WifiManager works perfectly fine. i.e I am able to ping the IP assigned dynamically, But for static IP, I tried both the programs in the example of WifiManager i.e AutoConnectwithStaticIp.ino and also AutoConnectwithFSparametersandCustomIp.ino.
But using both the above examples when I observe in the serial monitor, I see its connected and the IP i have assigned also shows connected, but when I try to ping that IP it does not work. Also on restarting the nodemcu,  its again going in AP mode

Please anyone having a solution to this problem, I have been working on this for almost a week without success

Thanks

RmenaLopez, 2019-07-31:
Hi! I had se same problem. Please use the development branch, this branch has the static IP problem fixed, as well as some UI changes. Good Luck!

tablatronix, 2019-07-31:
I know there was a staticip issue with reconnects in master, it might also be fixed in hotfixes, I forget