tzapu/WiFiManager #626: sketch_jun18a:11: error: 'wifiManager' does not name a type

nufuk, 2018-06-18:
While trying the example code I get the following error when verifing:

`sketch_jun18a:11: error: 'wifiManager' does not name a type`

But I just literally copy & pasted the example from the README.md

tablatronix, 2018-06-18:
do you have WiFiManager wifiManager in head?

tablatronix, 2018-06-19:
Why? Just use the examples included in the library. The readme is probably wrong

tablatronix, 2018-06-19:
Not sure whats wrong with it, looks fine to me

closed 2018-06-19 by tablatronix

wdmtech, 2020-10-08:
It's because you need the other dependencies that aren't listed - not that easy to figure it out as a beginner (unless you go straight to running the examples). 

I'd PR but this project is new to me since this evening so I wouldn't want to touch anything yet!