tzapu/WiFiManager #630: how to use labelPlacement?

ideaChenGo, 2018-06-22:
WiFiManagerParameter(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom, int labelPlacement);

I don't know how to use labelPlacement,\
Can you help me please?
Anyway, thanks

tablatronix, 2018-06-23:
`WFM_LABEL_BEFORE`
`WFM_LABEL_AFTER`
`WFM_NO_LABEL`

dontsovcmc, 2018-06-28:
We couldn’t find any code matching 'WFM_LABEL_BEFORE' in tzapu/WiFiManager

tablatronix, 2018-06-28:
Development branch

dontsovcmc, 2018-06-28:
oh.. thanks. Do you know new release date? 
Maybe you should fork repository and make your own releases? Sorry, I don't know, why author stopped releasing since 2016.

tablatronix, 2018-06-28:
development branch is actively updated.. Alpha should be soon

dontsovcmc, 2018-07-06:
@tablatronix remove init function without labelPlacement from header as no used:
```
void init(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom);
```
I can create pull request, if needed.

offtopic:
I use custom parameters in my project. Maybe it's better to add them somewhere? 

```
class IPAddressParameter : public WiFiManagerParameter {
public:

    IPAddressParameter(const char *id, const char *placeholder, IPAddress address)
        : WiFiManagerParameter("")
    {
        init(id, placeholder, address.toString().c_str(), 16, "");
    }

    IPAddress getValue() {
        IPAddress ip;
        ip.fromString(WiFiManagerParameter::getValue());
        return ip;
    }
};

class LongParameter : public WiFiManagerParameter {
public:

    LongParameter(const char *id, const char *placeholder, long value, const uint8_t length = 10)
        : WiFiManagerParameter("") {

        init(id, placeholder, String(value).c_str(), length, "");
    }

    long getValue() {
        return String(WiFiManagerParameter::getValue()).toInt();
    }
};

class FloatParameter : public WiFiManagerParameter {
public:

    FloatParameter(const char *id, const char *placeholder, float value, const uint8_t length = 10)
        : WiFiManagerParameter("") {
            
        init(id, placeholder, String(value).c_str(), length, "");
    }

    float getValue() {
        return String(WiFiManagerParameter::getValue()).toFloat();
    }
};
```

fthiebolt, 2018-11-16:
Hello,

I have had used Ken Taylor version for some time but since dev. branch also features add on parameters, i'm switching back ... with issues:

`
const char _customHtml_checkbox[]         = "type=\"checkbox\"";
WiFiManagerParameter p_sandbox("sandbox", "neOCampus sandbox", "T", 2, _customHtml_checkbox, WFM_LABEL_AFTER);
`
but there's a misalignment as you can see in attached picture ... we want to have both checkbox + LMabel on same line ... my mistake ??

[image]

Thanks
François

tablatronix, 2018-11-16:
hmm Ill take a look

Blueforcer, 2019-08-14:
Any news about the misalignment?

tablatronix, 2019-08-14:
I have not looked, sorry forgot about this. Will take a look now

tablatronix, 2019-08-14:
Might be an android thing??
Iphone
[image]

Blueforcer, 2019-08-21:
its similar to android, should be the checkbox in one line with the text? In your screenshot the checkbox is also above the text

tablatronix, 2019-08-21:
Shrug then you have more space for chars without wrapping or adding word break rules. Styling checkboxes is a pita, this should be using default browser styles, feel free to make css suggestions..