tzapu/WiFiManager #634: Setting configuration from Wifi Manager (SSID, Password, IP, GW, Subnet, dns, etc)

adamrabbani, 2018-06-28:
Hello @tablatronix 

i want know what variable i will use..
if ESP disconnected from wifi, i want call Wifi.begin with SSID and Password in "loop" from setting configuration in WifiManager.. and other variable like (IP, GW, Subnet, dns, etc)

Thanks

tablatronix, 2018-06-28:
You just call begin this is esp8266 related not wifimanager

adamrabbani, 2018-06-28:
@tablatronix 
just call WiFi.begin()? without variable? 
because if ESP accidentally disconnect from WiFi, i want using setting configuration from wifimanager.. 
i dont want reset the ESP and configuration again... 
Thanks

tablatronix, 2018-06-28:
They are one in the same

tablatronix, 2018-06-28:
Esp automatically reconnects to wifi on its own

adamrabbani, 2018-06-28:
@tablatronix 
Thank you for clarifying that.

closed 2018-06-28 by adamrabbani