tzapu/WiFiManager #635: I want to add a separate translation button ESP8266 in the configuration mode

DeltaVetal26, 2018-06-28:
Hi, I plan to use WifiManager in my project, now I'm in the training phase of WifiManager. I have an ESP8266, I want to make sure that the first time ESP8266 was loaded in the settings mode, where the user enters data from the Wi-Fi network (this is implemented in WifiManager), then the device will connect to this network every time it starts and works. But if suddenly the connection with the network disappears, as I understand it, the ESP8266 with WifiManager on board will be transferred to the configuration mode. There are times when ESP8266 is started, and the WIfi network is unexpectedly disconnected indefinitely. The question is, is it possible to make sure that when the network fails, the ESP8266 is not transferred to the configuration mode, but just waiting to be connected to the network?
Is it possible to add a separate button to the ESP pins, when pressing which ESP will be translated in the configuration mode? ESP is far from me, so I would like to prevent the unauthorized transfer of ESP in configuration mode.

tablatronix, 2018-06-28:
Wifimanager only enters config when you tell it.
It does not automatically start.

DeltaVetal26, 2018-06-29:
@tablatronix 
It is very cool! Many thanks for your help!

closed 2018-07-20 by tablatronix