tzapu/WiFiManager #638: Is it possible for user to know if ssid and password input correctly?

kotran88, 2018-06-29:
Hello :) I use this library well. 

but want to know that on /wifisave page, 
I want user to know if they enter ssid and password correctly, 

for now, 
when users enter ssid / password on /wifi page, 
it redirect to page according to below code no matter password is correct or incorrect. 
but I want to alert user if password is incorrect. 

is it possible? 
On connectWifi function, when connRes is 4 , it is password incorrect . I wonder it can trigger alert or alert page. and how. 

server->sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
  server->sendHeader("Pragma", "no-cache");
  server->sendHeader("Expires", "-1");
  server->setContentLength(CONTENT_LENGTH_UNKNOWN);
  server->send(200, "text/html", ""); // Empty content inhibits Content-length header so we have to close the socket ourselves.

  String head = FPSTR(HTTP_HEAD);
  head.replace("{v}", "Credentialsdddddd Saved");
  server->sendContent(head);
  server->sendContent_P(HTTP_SCRIPT);
  server->sendContent_P(HTTP_STYLE);
  server->sendContent_P(HTTP_HEAD_END);

  server->sendContent_P(HTTP_SAVED);

  server->sendContent_P(HTTP_END);
  server->client().stop();

tablatronix, 2018-06-29:
Not at this time, exploring options

rohitjust24idpl, 2019-02-25:
@tzapu @tablatronix 
Any update on this? Because if user enters ssid/psk incorrectly by mistake, it goes into a mode where user gets confused what to do ahead and they are not aware if they entered it correctly or not.

tablatronix, 2019-12-04:
#614

closed 2019-12-04 by tablatronix