tzapu/WiFiManager #643: Need to connect through an API rather WebView

solankigithub, 2018-07-06:
Hello,
I would like to connect and provide the Wifi credentials using a Web Api rather then a Web View.
I would like to call a URL (for eg: http://192.168.4.1/connect?ssid=wifissid&password=12345678)

I need this so that i can provide the credential through my anrdoid app. I will call the above URL from android app after connecting to AP.

Please let me know if there is a way to achieve it.

Thanks
Sandeep

tablatronix, 2018-07-06:
submit a PR

tablatronix, 2018-07-06:
This is on roadmap for future, but you can easily hack in a json response if querystring contains json or somehting

tarontop, 2021-12-02:
This is exactly what I'm looking for, is there any progress?

tablatronix, 2021-12-03:
no but you can add your own endpoint callbacks now since the webserver is a shared pointer, see the examples