tzapu/WiFiManager #644: incorrect cyrillic when reopen Portal (if connect to router failed)

dontsovcmc, 2018-07-06:
### Basic Infos
#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			esp01

Core Version:      	2.4.0, staging

### Description

WifiManager turns on AP for setup
I input settings, push OK

```
*WM: title: Новые показания {DEVICE_NAME}
*WM: template: ГВС: {V0} м3, ХВС: {V1} м3\r\nдельта:\r\nгвс: +{V3}, хвс: +{V4}\r\nпитание:{V2} В\r\nCMC:\r\nвода добавить: {V0} {V1}
```

ESP8266 connect to my WiFi failed
I reopen Settings page and write correct WiFi settings
I see  incorrect cyrillic encoding...
```
*WM: title: &#1053;&#1086;&#1074;&#1099;&#1077; &#1087;&#1086;&#1082;&#1072;&#1079;&#1072;&#1085;&#1080;&#1103; {DEVICE_NAME}
*WM: template: &#1043;&#1042;&#1057;: {V0} &#1084;3, &#1061;&#1042;&#1057;: {V1} &#1084;3\r\n&#1076;&#1077;&#1083;&#1100;&#1090;&#1072;:\r\n&#1075;&#1074;&#1089;: +{V3}, &#1093;&#1074;&#1089;: +{V4}\r\n&#1087;&#1080;&#1090;&#1072;&#1085;&#1080;&#1077;:{V2} &#1042;\r\nCMC:\r\n&#1074;&#1086;&#1076;&#1072; &#1076;&#1086;&#1073;&#1072;&#1074;&#1080;&#1090;&#1100;: {V0} {V1}
```

### Settings in IDE
platformio.ini
```
platform = espressif8266
framework = arduino
upload_port = COM7 ;/dev/tty.SLAB_USBtoUART  ; COM6

lib_deps =
  Blynk@0.5.2
  WifiManager=https://github.com/dontsovcmc/WiFiManager.git#waterius
```
### Source

https://github.com/dontsovcmc/waterius/blob/dev/ImpCounterESP/src/setup_ap.cpp#L57

### Debug Messages

```
␀l��|␀�l�|␂␌␌␌�␌l�␌b|��␂�␒�r�bl␌b��nn�lnn���␌b␜p␌lb␎lrlp�n�␐␂␌␌�␌l␌��␌␌␌b␌n�|␂l�␌␌�b��nn�␀l��l`␂�␒␒nn␌l`␂␎␂nr���n␌␌b␌�␎l␎r��n␌␌b␌�␎l��l␌���p��l`␂��n�␂000:00:00:00:051  NOTICE    (ESP) : Booted
1 000:00:00:00:053  NOTICE    (I2C) : mode=1
3 1 D3 B 0 0 4 0 0 0 0 0 0 0 000:00:00:00:065  NOTICE    (I2C) : data received
000:00:00:00:065  NOTICE    (WIF) : crc failed=1236
000:00:00:00:066  NOTICE    (WIF) : Init config: IP=192.168.0.116, Subnet=255.255.255.0, Gw=192.168.0.1, hostname=blynk-cloud.com
000:00:00:00:075  NOTICE    (ESP) : new impulses=4 0
000:00:00:00:079  NOTICE    (ESP) : new values=0.00 0.00
000:00:00:00:084  NOTICE    (ESP) : I2C-begined: mode SETUP
000:00:00:00:090  NOTICE    (WIF) : User requested captive portal
*WM: Added Parameter: ip
*WM: Added Parameter: subnet
*WM: Added Parameter: gw
*WM: Added Parameter: key
*WM: Added Parameter: host
*WM: Added Parameter: email
*WM: Added Parameter: title
*WM: Added Parameter: template
*WM: Added Parameter: value0
*WM: Added Parameter: value1
*WM: Added Parameter: factor
*WM: Disabling STA
*WM: Enabling AP
*WM: StartAP with SSID:  Waterius_0.4.1
*WM: AP has anonymous access!
*WM: AP IP address: 192.168.4.1
*WM: Starting Web Portal
*WM: HTTP server started
*WM: WiFi Scan done
*WM: Config Portal Running, blocking, waiting for clients...
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: Scan is cached
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- HTTP Info
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: Portal Timeout In 279 seconds
*WM: <- HTTP Wifi
*WM: Scan is cached
*WM: 9 networks found
*WM: DUP AP: newsprin
*WM: AP:  -63 a2-tst
*WM: AP:  -63 alpha_2
*WM: AP:  -72 AlphaGuest
*WM: AP:  -74 newsprin
*WM: AP:  -84 YOTA-3F7E
*WM: AP:  -84 DIRECT-5xKM2FDDA3
*WM: AP:  -87 Bunker
*WM: AP:  -88 hpsetup
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: Portal Timeout In 277 seconds
*WM: <- HTTP WiFi save
*WM: Parameters
*WM: --------------------
*WM: ip: 10.0.24.171
*WM: subnet: 255.255.255.0
*WM: gw: 10.0.24.1
*WM: key: xxxx
*WM: host: blynk-cloud.com
*WM: email: xxxx
*WM: title: Новые показания {DEVICE_NAME}
*WM: template: ГВС: {V0} м3, ХВС: {V1} м3\r\nдельта:\r\nгвс: +{V3}, хвс: +{V4}\r\nпитание:{V2} В\r\nCMC:\r\nвода добавить: {V0} {V1}
*WM: value0: 0.00
*WM: value1: 0.00
*WM: factor: 10
*WM: --------------------
*WM: process connect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: [ERROR] wifi config failed
*WM: Connecting to new AP: a2-tst
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECT_FAILED
*WM: [ERROR] Connect to new AP Failed
*WM: Disabling STA
*WM: Portal Timeout In 280 seconds
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: WiFi Scan done
*WM: <- HTTP Wifi
*WM: Scan is cached
*WM: 9 networks found
*WM: DUP AP: newsprin
*WM: AP:  -61 a2-tst
*WM: AP:  -62 alpha_2
*WM: AP:  -73 newsprin
*WM: AP:  -75 AlphaGuest
*WM: AP:  -82 DIRECT-5xKM2FDDA3
*WM: AP:  -91 kaskade-2
*WM: AP:  -92 hpsetup
*WM: AP:  -92 Office
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: <- Request redirected to captive portal
*WM: Portal Timeout In 280 seconds
*WM: <- HTTP WiFi save
*WM: Parameters
*WM: --------------------
*WM: ip: 10.0.24.172
*WM: subnet: 255.255.255.0
*WM: gw: 10.0.24.1
*WM: key: xxxx
*WM: host: blynk-cloud.com
*WM: email: xxxxx
*WM: title: &#1053;&#1086;&#1074;&#1099;&#1077; &#1087;&#1086;&#1082;&#1072;&#1079;&#1072;&#1085;&#1080;&#1103; {DEVICE_NAME}
*WM: template: &#1043;&#1042;&#1057;: {V0} &#1084;3, &#1061;&#1042;&#1057;: {V1} &#1084;3\r\n&#1076;&#1077;&#1083;&#1100;&#1090;&#1072;:\r\n&#1075;&#1074;&#1089;: +{V3}, &#1093;&#1074;&#1089;: +{V4}\r\n&#1087;&#1080;&#1090;&#1072;&#1085;&#1080;&#1077;:{V2} &#1042;\r\nCMC:\r\n&#1074;&#1086;&#1076;&#1072; &#1076;&#1086;&#1073;&#1072;&#1074;&#1080;&#1090;&#1100;: {V0} {V1}
*WM: value0: 0.00
*WM: value1: 0.00
*WM: factor: 10
*WM: --------------------
*WM: process connect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: STA IP set: 10.0.24.172
*WM: Connecting to saved AP: a2-tst
*WM: 10000 ms connectTimeout set
*WM: 10000 ms timeout, waiting for connect...
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: .
*WM: Connection result: WL_CONNECTED
*WM: Connect to new AP [SUCCESS]
*WM: Got IP Address:
*WM: 10.0.24.172
*WM: disconnect configportal
*WM: config portal exiting
000:00:02:22:129  NOTICE    (WIF) : Connected to wifi
000:00:02:22:134  NOTICE    (DAT) : impulses0_start=data.impulses0=4
000:00:02:22:140  NOTICE    (DAT) : impulses1_start=data.impulses1=0
000:00:02:22:185  NOTICE    (WIF) : Config stored: IP=10.0.24.172, Subnet=255.255.255.0, Gw=10.0.24.1, hostname=blynk-cloud.com
000:00:02:22:186  NOTICE    (WIF) : key=xxx
000:00:02:22:192  NOTICE    (WIF) : value0_start=0.00, impules0_start=4, factor=10
000:00:02:22:199  NOTICE    (WIF) : value1_start=0.00, impules1_start=0
000:00:02:22:206  NOTICE    (ESP) : Going to sleep

```

tablatronix, 2018-07-20:
I think the webserver is doing this. hmm

tablatronix, 2018-07-20:
It looks like its get encoded, then gets cut off because of param length.

tablatronix, 2018-07-20:
Nope webbrowser encodes on post I think

tablatronix, 2018-07-20:
Maybe if we add charset=utf-8 header.. ill check

dontsovcmc, 2018-07-21:
I can't understand the reason: sometimes it opens correctly, sometimes wrong.
It also happenes in the first use.

dontsovcmc, 2018-08-04:
Mobile phone 
This characters `&#1053;&#1086;&#1074;&#1099;&#1077;` is WINDOWS-1251 encoding. There are 4 bytes for 1 character needed...

Add charset="UTF-8" and see correct utf-8 bytes:
d0 9d d0 be d0 b2 d1 8b d0 b5 20 d0 b7 d0 bd d0 b0 d1 87 d0 b5 d0 bd d0 b8 d1 8f 20

tablatronix, 2018-08-10:
@dontsovcmc can we have this tested and closed

tablatronix, 2018-08-10:
I think the param ID will have to be lower ascii, because it is checked for invalid characters

dontsovcmc, 2018-08-11:
@tablatronix, i saw this document : https://www.w3schools.com/tags/att_meta_charset.asp

I check it only in chrome.

tablatronix, 2018-08-11:
I can test some more but can we close this now ?

dontsovcmc, 2018-08-11:
of course

closed 2018-09-21 by tablatronix