tzapu/WiFiManager #648: "AutoConnectAp" name comebacks after disconnect.

hassasian, 2018-07-09:
Hi. I'm using esp8266 with wm ofc. Everthing works as expected thx for this great lib. But I encounter a problem today. Here we go  ; 
On setup,
```
 if (!wifiManager.autoConnect("BlaBla")) {
    Serial.println("failed to connect and hit timeout");
  }

```
It works as expected at the begining. I connect my esp to a router. Then I close the router. And esp disconnect from it normally. Then it starts ap mode which named "AutoConnectAp".  But I want same name for this ap mode.

tablatronix, 2018-07-09:
You will have to erase credentials, Sometimes it gets saved to flash

hassasian, 2018-07-09:
Thx for your answer. But i dont understad what u mean. If i delete credentials will that cause to loose my ip configs etc. And how i delete this?

hassasian, 2018-07-10:
thx for helping my problem to solve 👍 
    `wifiManager.resetSettings();`
solved it.

closed 2018-07-10 by hassasian