tzapu/WiFiManager #650: Compilation of example AutoConnect fail for NodeMCU 1.0

naveenneog, 2018-07-10:
I used the latest Arduino/ESP8266 core and trying to build the examples for the NodeMCU 1.0. 
I am using Arduino 1.8.5 IDE. The compilation is failing with below errors. I used the Development branch also to see any new fixes , but same issue repeats. 
```
/var/folders/jb/0x94y7b91czbys0plp0f9gwc0000gn/T/arduino_build_221592/libraries/WiFiManager/WiFiManager.cpp.o: In function `String::operator!=(char const*) const':
Multiple libraries were found for "WiFiManager.h"
/Users/neo-optimus/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:712: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
 Used: /Users/neo-optimus/Documents/Arduino/libraries/WiFiManager
 Not used: /Users/neo-optimus/Documents/Arduino/libraries/WiFiManager-master
/var/folders/jb/0x94y7b91czbys0plp0f9gwc0000gn/T/arduino_build_221592/libraries/WiFiManager/WiFiManager.cpp.o: In function `WiFiManager::startWPS()':
/Users/neo-optimus/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:712: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'
collect2: error: ld returned 1 exit status
Using library WiFiManager at version 0.12 in folder: /Users/neo-optimus/Documents/Arduino/libraries/WiFiManager 
Using library ESP8266WiFi at version 1.0 in folder: /Users/neo-optimus/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: /Users/neo-optimus/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: /Users/neo-optimus/Documents/Arduino/hardware/esp8266com/esp8266/libraries/DNSServer 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

```

d-a-v, 2018-07-10:
Until this is fixed, you can enable WPS in the core:
```
./tools/boards.txt.py --allowWPS --allgen
```

liebman, 2018-07-10:
This was fixed in the development branch with this PR https://github.com/tzapu/WiFiManager/pull/645 3 days ago.

tablatronix, 2018-07-10:
I have to commit this to hotfixes

naveenneog, 2018-07-11:
@liebman  I tried the development branch too It doesn't fixed the issue. but @d-a-v fix helped me to continue . Thank you

liebman, 2018-07-11:
@naveenneog what were the error messages when you used the development branch? I want to make sure I did not miss something in the fix for WiFiManager.  My current assumption is that something in your sketch or in another library was referencing the WPS functionality, in which case, @d-a-v's sugestion would be correct.

naveenneog, 2018-07-15:
You would see the same error message I posted in my original issue. @d-a-v suggestion kept me going so closing this issue

closed 2018-07-15 by naveenneog

akshaykankal, 2018-10-10:
I solved it by just downgrading the arduino json library version. It worked.

Craig1516, 2019-04-06:
@d-a-v .  Could you please elaborate where your fix is applied?  I put this in the File/Preferences/Additional Boards Manager box and got a series of other errors.  I must be doing this wrong.

d-a-v, 2019-04-06:
@Craig1516 I think you should update your versions.
I think this bug is only happening with esp8266-arduino-core version 2.4.1 (we have 2.4.2 and 2.5.0 now).

Craig1516, 2019-04-07:
@d-a-v  Thanks!  I upgraded my Arduino IDE from 1.8.1 to 1.8.9 and that fixed this problem.  But now my SPIFFS Sketch Data Upload menu selection is gone and I'm researching how to recover.  Two steps forward, and one back.

d-a-v, 2019-04-07:
Just take it from your 1.8.1 `Tools` sub directory (ESP8266FS).

Craig1516, 2019-04-08:
@d-a-v   Thanks again.  If I were a smart fellow, I would have saved that information - but for the upgrade.  Fortunately, I found the *.jar file from you and another fellow and I copied that into the tool(s) directory(ies).   The forks in the road have come together.  My other inquiry/answers are @https://github.com/esp8266/Arduino/issues/5936#issuecomment-480563084  for those reading.  Hope this helps others.  In that blog I ask, maybe Sketch Data Upload is obsolete?  My file was able to create the SPIFFS file that I wanted - unexpectedly!