tzapu/WiFiManager #652: How to don't be blocked by wifiManager.autoConnect()?

spilz87, 2018-07-10:
Hello, 
I'm using wifiManger library and it works very well for me.
But for my projects, my esp8266s have other stuff to do than wifi stuff (like receive data through serial port and display them on screen) independently of wifi connection.
So I need a way to don't be blocked during wifi connection, more over when there is no wifi in house. 
Is it possible to have like a state machine for wifi connection that is call in loop() but doesn't block the use of the esp ?
Sorry for my bad English, I hope you understand what I mean, if no, let me know to try explain differently 

Thanks for your help

d-a-v, 2018-07-10:
I have the same request.
For my personal use I added a callback for that purpose: [offline-loop](https://github.com/tzapu/WiFiManager/compare/master...d-a-v:loop)
I did not make it a PR. Devs might consider implementing a similar feature for working-offline guys like us :)

tablatronix, 2018-07-11:
Develoment branch setconfigportalblocking false , process() in loop

spilz87, 2018-07-11:
That's the idea, I prefer the way to have a process in loop than to have to separe online and offline part

tablatronix, 2018-07-11:
I am making an example now, It has not been tested yet though.

something like this should work

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("AutoConnectAP")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
}

void loop() {
    wm.process();
    // put your main code here, to run repeatedly:
}
```

d-a-v, 2018-07-11:
Thanks a lot !

tablatronix, 2018-07-14:
dup #379

closed 2018-07-14 by tablatronix

spilz87, 2018-07-15:
Hello,
Actually it doesn't really solve all my request :
I need no blocking at all, even during connection.
If I understand well with your exemple, we don't stay blocked if it run portail, but if there is wrong wifi settings, it stay on the conne ruin loop, until connection time out (if set).
For my projects with an oled screen and rotary encoder (like on 3d printer) I need the ESP react to the encoder, connected or trying to connect or not connected. Like a state machine run in loop()

tablatronix, 2018-07-15:
Then don't use autoconnect or use a really short connecttimeout.
Not sure what to tell you more specifically.

spilz87, 2018-07-16:
you mean by using begin instead of autoconnect ?

tablatronix, 2018-07-16:
Yeah and manually check connect status and start configportal in your code so you have better control.