tzapu/WiFiManager #654: ERROR when tried to complie  'DynamicJsonBuffer' was not declared in this scope

Ameengray, 2018-07-12:
.I get the following error when i try to compile this code, can this be fixed its urgent, please lookinto it. 

Arduino: 1.8.5 (Linux), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

/root/Arduino/Blink2/Blink2.ino: In function 'void setup()':
Blink2:51: error: 'DynamicJsonBuffer' was not declared in this scope
         DynamicJsonBuffer jsonBuffer;
         ^
/root/Arduino/Blink2/Blink2.ino:51:9: note: suggested alternative:
In file included from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /root/Arduino/Blink2/Blink2.ino:10:
/root/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
Blink2:51: error: expected ';' before 'jsonBuffer'
         DynamicJsonBuffer jsonBuffer;
                           ^
Blink2:52: error: 'jsonBuffer' was not declared in this scope
         JsonObject& json = jsonBuffer.parseObject(buf.get());
                            ^
Blink2:53: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
         json.printTo(Serial);
              ^
Blink2:130: error: 'DynamicJsonBuffer' was not declared in this scope
     DynamicJsonBuffer jsonBuffer;
     ^
/root/Arduino/Blink2/Blink2.ino:130:5: note: suggested alternative:
In file included from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonDocument.hpp:10:0,
                 from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /root/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /root/Arduino/Blink2/Blink2.ino:10:
/root/Arduino/libraries/ArduinoJson/src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
Blink2:130: error: expected ';' before 'jsonBuffer'
     DynamicJsonBuffer jsonBuffer;
                       ^
Blink2:131: error: 'jsonBuffer' was not declared in this scope
     JsonObject& json = jsonBuffer.createObject();
                        ^
Blink2:141: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
     json.printTo(Serial);
          ^
Blink2:142: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'
     json.printTo(configFile);
          ^
Multiple libraries were found for "ArduinoJson.h"
 Used: /root/Arduino/libraries/ArduinoJson
 Not used: /root/Arduino/libraries/ArduinoJson-master
exit status 1
'DynamicJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tablatronix, 2018-07-12:
what does this have to do with wifimanager?

Ameengray, 2018-07-12:
This the given code https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithFSParameters 

When i want to be able to send custom parameters through the wifi manager, like mqtt username and password, when i refer to link provided in the documentation and compile the program, i get the above error, can you please why i am getting that error, i get the same error with other examples too.

Ameengray, 2018-07-12:
I got the answer, The problem was with the ArduinoJson, the version 6 is the beta stage, use version 5x to fix this issue.

tablatronix, 2018-07-12:
ok I saw blink.ino nothing about wm.
I guess the examples will have to be updated at some point it seems.

per1234, 2018-07-12:
Some info:
- https://arduinojson.org/v5/faq/error-jsonbuffer-was-not-declared-in-this-scope/
- https://arduinojson.org/v6/doc/upgrade/

saubeo6600, 2018-07-18:
please fix this code for arduino ide 1.6.4 and aduinojson 5.13.2
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[6] = "8080";
char blynk_token[34] = "YOUR_BLYNK_TOKEN";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(blynk_token, json["blynk_token"]);

        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_blynk_token("blynk", "blynk token", blynk_token, 32);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  wifiManager.setSTAStaticIPConfig(IPAddress(10,0,1,99), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
  
  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_blynk_token);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();
  
  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(blynk_token, custom_blynk_token.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");

    JsonObject& json = jsonBuffer.createObject();
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["blynk_token"] = blynk_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.printTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());

}

void loop() {
  // put your main code here, to run repeatedly:

}

saubeo6600, 2018-07-18:
Thanks for the feedback soon!
Source code I coppy from the link: https: //github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino
I really need it, but I can not fix error:
    **DynamicJsonBuffer easiest declared in this scope**

: Arduino: 1.6.4 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80MHz, 4M (1M SPIFFS), v2 Lower Memory,

esp8266_final.ino: In function 'void setup ()':
esp8266_final: 51: error: 'DynamicJsonBuffer' was not declared in this scope
esp8266_final.ino: 51: 9: note: suggested alternative:
ArduinoJson / DynamicJsonDocument.hpp: 10: 0,
                 from: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ arduino_173318 \ src / ArduinoJson.hpp: 9,
                 from ArduinoJson.h: 9,
                 from esp8266_final.ino: 10:
ArduinoJson :: Internals :: DynamicJsonBuffer ':: ArduinoJson :: Internals :: DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
esp8266_final: 51: error: expected ';' trước khi jsonBuffer
esp8266_final: 52: error: 'jsonBuffer' was not declared in this scope
error: 'class ArduinoJson :: JsonObject' has no member named 'printTo'
esp8266_final: 54: error: 'class ArduinoJson :: JsonObject' has no member named 'success'
esp8266_final: 130: error: 'DynamicJsonBuffer' was not declared in this scope
esp8266_final.ino: 130: 5: note: suggested alternative:
ArduinoJson / DynamicJsonDocument.hpp: 10: 0,
                 from: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ arduino_173318 \ src / ArduinoJson.hpp: 9,
                 from ArduinoJson.h: 9,
                 from esp8266_final.ino: 10:
ArduinoJson :: Internals :: DynamicJsonBuffer ':: ArduinoJson :: Internals :: DynamicJsonBuffer'
     DynamicJsonBuffer;
     ^
esp8266_final: 130: error: expected ';' trước khi jsonBuffer
esp8266_final: 131: error: 'jsonBuffer' was not declared in this scope
esp8266_final: 141: error: 'class ArduinoJson :: JsonObject' has no member named 'printTo'
esp8266_final: 142: error: 'class ArduinoJson :: JsonObject' has no member named 'printTo'
Multiple libraries were found for "WiFiManager.h"

 Used: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ WiFiManager-master

 Not used: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ WhareHauoraWiFiManager

Libraries were found for "ArduinoJson.h"

 Used: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ arduino_173318

 Not used: C: \ Users \ Administrator \ Documents \ Arduino \ libraries \ ArduinoJson

'DynamicJsonBuffer' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  được bật trong File> Preferences.

tablatronix, 2018-07-19:
Ideally we would detect which version and handle, although this is just an example and not part of the library

brentru, 2018-08-17:
@tablatronix Heya - I'm planning to use this library for the Adafruit IO Arduino Library (https://github.com/adafruit/Adafruit_IO_Arduino) and would be happy to work on this issue. Could you assign it to me?

tablatronix, 2018-08-17:
Not atm, but you can PR it against development branch

brentru, 2018-08-17:
Thanks - will do!

tablatronix, 2018-08-17:
Maybe add support in the example in question for both versioons somehow?
Is it still in beta? Maybe make a branch for next version of arduinojson 6 or whatever

brentru, 2018-08-17:
The ArduinoJSON library is still in beta 6.x. I'm going to source the version in the example (if they expose a version) and handle the error.

LiamGomes27, 2019-01-15:
Guys the code is still not compiling for me even after I downgrade the version on arduinojson library. Can anyone help me out?  I tried version 5.12.1, 5.12.2 and even 5.13

Jewel852, 2019-09-08:
Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\srjew\Documents\Arduino\libraries\arduino-youtube-api-master\src\YoutubeApi.cpp:95:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

  DynamicJsonBuffer jsonBuffer;

           ^

C:\Users\srjew\Documents\Arduino\libraries\arduino-youtube-api-master\src\YoutubeApi.cpp: In member function 'bool YoutubeApi::getChannelStatistics(String)':

C:\Users\srjew\Documents\Arduino\libraries\arduino-youtube-api-master\src\YoutubeApi.cpp:95:20: error: 'jsonBuffer' was not declared in this scope

  DynamicJsonBuffer jsonBuffer;

                    ^

C:\Users\srjew\Documents\Arduino\libraries\arduino-youtube-api-master\src\YoutubeApi.cpp:97:10: error: 'ArduinoJson::JsonObject' has no member named 'success'

  if(root.success()) {

          ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tablatronix, 2019-09-08:
I have to fix this, I will try to do it this week

tihoangyeudau, 2019-09-08:
> Arduino: 1.8.9 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200" [...]

you need to downgrade arduinoJson to v5.11.3

per1234, 2019-09-08:
>I pushed v0.15-beta but it is not showing in arduino library manager still, no idea wtf is the deal with arduino

@tablatronix I don't know for sure because I have no access to the library indexer script, but I think the problem must be the `version` value in library.properties for that release: 0.15-beta. Although the indexer does support non-semver versions like 0.15, I don't know that it supports the prerelease syntax on top of the non-semver version.

What is certain is that any semver-compliant version will be accepted. So my recommendation is to just start using semver-compliant versioning:
https://semver.org/

Spackstor, 2020-01-21:
Try adding a size to            DynamicJsonDocument doc(1024);          worked for me.
Go here for a little guidance.       

> https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/

Ruandv, 2020-01-27:
@tablatronix  this can be closed because of pull request #993

closed 2020-01-27 by tablatronix