tzapu/WiFiManager #655: ESP32 development branch - 2 fixes and problem accessing OnDemandAP config page

Mirdox, 2018-07-13:
WiFimanager Branch/Release:  Development

Hello there,

Thanks for this project, it works perfectly on esp8266.
Now I was trying on esp32 and I'm facing several issues.
Some of them i was able to fix.

Fix 1: There is conflict with esp32 library WebServer file HTTP_Method.h which uses same variable HTTP_HEAD. Renaming it in WiFiManager.cpp and strings_en.h to HTML_HEAD solves the problem.
HTML_HEAD is btw more appropriate, because it is HTML Header, not HTTP.

Fix 2: There is a typo in WiFiManager.cpp line 1081
bool tok_I = HTTP_PARAM_temp.indexOf(FPSTR(T_i)) > 0; // lower case i
should be changed to
bool tok_I = HTTP_PARAM_temp.indexOf(FPSTR(T_I)) > 0; // upper case I

Current problem: now I'm stuck, because I cannot open config page in browser while connected to OnDemandAP. When the Wifi Credentials are missing the config page is working fine.
There is no crash and no debug messages and I've tried on more devices.

Any idea?

Thank you

tablatronix, 2018-07-13:
I just used it he other day, were there alot of commits to esp32?

tablatronix, 2018-07-13:
Ah i see webserver abstractions, I just looked for changes this morning and missed this. Will fix

commit 17f2316e (2018-07-13): fixes #655
commit 24d317b9 (2018-07-13): fix #655
closed 2018-07-13 by tablatronix