tzapu/WiFiManager #656: Accessing portal after connected to local wifi

vaz83, 2018-07-14:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware: NodeMCU ESP8266 Lua WiFi Internet Development Board

Core Version:      	2.4.0, staging

### Description

Hi, can anyone tell me how to access the webportal in runtime? I have seen some examples around here but to acess the webpage the user needs to push a button and then connect to the wifi created by the wifiManager. What i would need is after the wifiManager is configured and available on my local network, every time i enter its address on the web browser, it opens by default the configuration portal. Is is possible?

tablatronix, 2018-07-14:
Ondemandwebportal example

vaz83, 2018-07-14:
Hi tablatronix,
I have checked the example you said, but it's not what I want exactly. on that example, the user needs to trigger a button in order to activate the portal. What I am looking for, is that when the esp is connected to my local network, when I put it's IP address on the web browser, that it can open the portal by default. I have tried to create a webserver so that when i detect someone is trying to connect, i call the  wm.startWebPortal(), but it doesnt work that way. Any sugestions ?

tablatronix, 2018-07-14:
it is an example...
Then do not use a button, just do it whenever you want.

vaz83, 2018-07-14:
Yes, I know, but i am struggling to get the correct code, can I please ask you some little help on how can i do it?

tablatronix, 2018-07-14:
When do you want it to turn on? What is the logic

tablatronix, 2018-07-14:
works for me

```CPP
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

WiFiManager wm;
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "servername", 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", "1234", 6);

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    delay(3000);

    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around

    // add a custom input field

  WiFiManagerParameter custom_field("customfield", "Custom Field", "Default Value", 40);
  WiFiManagerParameter custom_token("api_token", "api token", "0123456789", 9); // overflow
  // WiFiManagerParameter custom_tokenb("invalid token", "invalid token", "abcdefghijk", 11);

  //add all your parameters here
  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);
  wm.addParameter(&custom_field);
  wm.addParameter(&custom_token);
  // wm.addParameter(&custom_tokenb); // overflow

    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);
    if(wm.autoConnect("AutoConnectAP")){
        //if you get here you have connected to the WiFi
        Serial.println("connected...yeey :)");
    }
    else 
    {
        Serial.println("non blocking config portal running");
    }

    wm.startConfigPortal("OnDemandAP");
}

void loop() {
    wm.process();
    // put your main code here, to run repeatedly:
    
}

```

vaz83, 2018-07-14:
I want to turn it on every time i access it's ip address on the local network

tablatronix, 2018-07-14:
So leave it on all the time?

vaz83, 2018-07-15:
Yes, and accessing the configuration portal through the local network and not by the auto generated wifi of the device

tablatronix, 2018-07-15:
so after your wifi connect code, or after checking to make sure wifi is connected however you are doing that just wm.startWebPortal, and wm.process() in your loop

vaz83, 2018-07-16:
done! Thanks man, you're awesome! I am sharing the code here so I can help someone who get's the same problem i did.

`#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;
char mqtt_server[40] = "0.0.0.0";
char mqtt_topicIN[80] ="IOT_Device/IN";
char mqtt_topicOUT[80] ="IOT_Device/OUT";
char mqtt_port[6]  = "1883";

WiFiManagerParameter custom_mqtt_server("server", "Server IP", mqtt_server, 40);
WiFiManagerParameter custom_mqtt_port("port", "Port", mqtt_port, 6);
WiFiManagerParameter custom_mqtt_topicIN("topicIn", "Input Topic", mqtt_topicIN, 80);
WiFiManagerParameter custom_mqtt_topicOUT("topicOut", "Output Topic", mqtt_topicOUT, 80);
WiFiManagerParameter custom_mqtt_messages("messages", "Messages Topic", mqtt_topicOUT, 80);

ESP8266WebServer server(80);

void handleRoot()
{
  wm.startWebPortal();
}
void setup()
{
  Serial.begin(115200);
  delay(3000);
  Serial.println("Setup mode...");
  //wifiManager.resetSettings();

  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);
  wm.addParameter(&custom_mqtt_topicIN);
  wm.addParameter(&custom_mqtt_topicOUT);
  wm.addParameter(&custom_mqtt_messages);

  wm.setConfigPortalBlocking(false);

  if(wm.autoConnect("IOT_Device"))
  {
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }
  else 
  {
    Serial.println("non blocking config portal running");
  }
  wm.startConfigPortal("IOT_Device"); 
  
  server.on("/", handleRoot);
  server.begin();
}

void loop()
{
  wm.process();
  server.handleClient();
  //Serial.println("hello world");
  // put your main code here, to run repeatedly:
}`

tablatronix, 2018-07-16:
startconfigportal already starts the webserver, you are starting it twice

tablatronix, 2018-07-16:
if you want to only start the webportal when the ap has a client you can use this.

`WiFi_softap_num_stations() > 0)`

vaz83, 2018-07-16:
changed it! thanks again :)

tablatronix, 2018-07-16:
So that handleroot is not necessary, also a few people have tried to run a webserver and wm at the same time and have not had much success

vaz83, 2018-07-16:
I am noticing now, that if I comment the startConfigPortal on the setup void, and keep the handleroot as it was, when i go to the device IP it does not open the wifimanager no more. is it asking too much if you could tell me what I need to change in the code? I just want to start the webportal when i acess the IP on the local network

tablatronix, 2018-07-16:
You cant detect "access the ip" unless you have a server running, so not exactly sure what you are trying to do or why.

You cannot do both, either start configportal(softap+webportal) or start webserver(webportal)

maybe the webportal wont start if you already have one running on port 80

vaz83, 2018-07-16:
I am trying to explain my best, my english is not that good i guess :) I just wanted that once the ESP is connected to the local network, i can access the webportal without the wifiManager launch the AP mode. When i call the startWebPortal(), the ESP is creating an AP, which I don't want. can i call the startWebPortal without the wifiManager creating a AP?

tablatronix, 2018-07-16:
startWebPortal should not start an AP, however the esp will start an AP if it got saved to flash.

Add `WiFi.mode(WIFI_STA)` after setup, in case its saved in memory  also erase flash )

vaz83, 2018-07-17:
It's now working perfectly, here's the code:
`#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

WiFiManager wm;
ESP8266WebServer server(80);

//############################################# CUSTOM PARAMETERS FOR THE WIFI MANAGER ##########################################
char mqtt_server[40] = "0.0.0.0";
char mqtt_topicIN[80] ="IOT_Device/IN";
char mqtt_topicOUT[80] ="IOT_Device/OUT";
char mqtt_port[6]  = "1883";

WiFiManagerParameter custom_mqtt_server("server", "Server IP", mqtt_server, 40);
WiFiManagerParameter custom_mqtt_port("port", "Port", mqtt_port, 6);
WiFiManagerParameter custom_mqtt_topicIN("topicIn", "Input Topic", mqtt_topicIN, 80);
WiFiManagerParameter custom_mqtt_topicOUT("topicOut", "Output Topic", mqtt_topicOUT, 80);
WiFiManagerParameter custom_mqtt_messages("messages", "Messages Topic", mqtt_topicOUT, 80);
//###############################################################################################################################

//################################################### GENERAL VARIABLES #########################################################
bool blockWM = true; // Change this to false if you want your code to continue to run on the loop void even if you are not conected to any wifi.
//###############################################################################################################################

void handleNotFound()
{
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++)
  {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message); 
}

/*
 * This void sets the device with the WiFiManager from tzapu with custom parameters.
 * Needs to be called only in the setup void.
*/
void setupDeviceWM()
{
  //wifiManager.resetSettings();
  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);
  wm.addParameter(&custom_mqtt_topicIN);
  wm.addParameter(&custom_mqtt_topicOUT);
  wm.addParameter(&custom_mqtt_messages);

  wm.setConfigPortalBlocking(blockWM);
  String ssid = "IOT_ESP_" + String(ESP.getChipId());
  
  if(wm.autoConnect(ssid.c_str()))
  {
    //if you get here you have connected to the WiFi
    Serial.println("Connected to wifi network!");
    WiFi.mode(WIFI_STA);
    wm.startWebPortal();
  }
  else 
  {
    Serial.println("Non blocking config portal running!");
  }

  // call the code down to activate wifi so users can configure the device, event if it's connected to the local network
  //wm.startConfigPortal("IOT_Device"); 
  //
  server.onNotFound(handleNotFound);
  server.begin(); // declare this at the beggining of the code => ESP8266WebServer server(80);
}

/*
 * This void needs to be called in the loop void so it can handle the WM and the webportal.
*/
void loopDeviceWM()
{
  wm.process();
  server.handleClient();
}

void setup()
{
  Serial.begin(115200);
  delay(3000);
  Serial.println("Setup mode...");
  //wifiManager.resetSettings();

  setupDeviceWM();
}

void loop()
{
  loopDeviceWM();
}`

closed 2018-07-17 by vaz83

SudhaMaanasa, 2020-06-20:
> It's now working perfectly, here's the code:
> `#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
>
> WiFiManager wm;
> ESP8266WebServer server(80); [...]

Could u please help me the same with my code
[WifiManagerWithWebserver.txt](https://github.com/tzapu/WiFiManager/files/4808064/WifiManagerWithWebserver.txt)
.

pioioTwo, 2020-08-17:
it is possible to put a password for access, example login acess portal?