tzapu/WiFiManager #657: Proposal for strategy to create in-device web apps accessible from mobile device home screen

jonathan-annett, 2018-07-14:
i've had some success on a small scale creating web apps that are served entirely within an esp8266 and are "launced" by tapping an icon on my phones home screen. this approach is fine for one-off deployment to "my phone" as i can hard code values as i see fit, but not really useful for deployment to third parties

I've been thinking wifimanager's startup logic could be massaged slightly to facilitate this approach, and perhaps a mini framework that launches a relatively useless starting point for more useful apps (for example the default functionality of the the "app" would be to allow the user to use their phone to toggle the LED on and off). once this project exists as a boilerplate framework, developers could replace this "hello world, i'm an LED app" to do what they want.

attached is starting flowchart of how the logic might be adapted to do this.

( there's another concept that would be simpler for the user, but requires them to edit the name of their phone's SSID to conform to a pattern based on their devices MAC, along with a preselected PSK based on that MAC (which they can later update to whatever they want, should the choose to)

[Untitled Diagram (2).pdf](https://github.com/tzapu/WiFiManager/files/2194600/Untitled.Diagram.2.pdf)

just wondering if you would consider working on  fork that takes this approach on board, or maybe incorporating ideas into the existing framework that would make this a function set of the whole project

tablatronix, 2018-07-14:
Once we figure out #461, this should be pretty easy for users to do without having to bloat the library.

and eventually we can add a spiffs webserver with some kind of webpack for making apps. I am just not convinced this needs to be a wm feature.

jonathan-annett, 2018-07-14:
the main issue is the user needs to be in a position where the phone is displaying the launch page on the correct ip. you can't just create a home page link with a "future ip address". since the phone is responsible for generating the ip address and serving it by dhcp. you can't guess at what that ip address might be.

jonathan-annett, 2018-07-14:
once allocated however. phone generally issue the same ip to connected MACs, assuming you use the app relatively regularly

jonathan-annett, 2018-07-14:
the PWA approach would work. however since it's a requirement for PWAs to have https, it's not easy to do that in a situation where the device doing the serving (the esp8266) corrrectly serve a valid cert tied to a valid dns entry, so we need to use plain http to an ip which "add to home screen" still supports.

tablatronix, 2018-07-14:
I still am not sure I understand how this solves this issue, the only way you can get to the server without an ip is captive portal(sotap) or sta with hostnames set ( and a compatible device for mdns )

jonathan-annett, 2018-07-14:
it's not a problem to save a home screen shortcut to an ip address. thats' not the hard part, it's communicating to the user what that address is, so they can display it in the first place.

tablatronix, 2018-07-14:
You cannot save a home screen from captive portal, the user had to manually goto the ip in a browser.
At least that I know of, am I wrong ?

jonathan-annett, 2018-07-14:
sure however if the captive portal serves some html with an A link to an ip address, the user clicks on that and they end up at a hard coded ip address in the address bar. they then add that url to the home screen.

jonathan-annett, 2018-07-14:
all "add to home screen" does is save a cached copy of the existing content, and the url that served it to a shutcut file. on reopen, it displays the cached content while it attempts to reload the url. it then updates the browser with actual content.

there are some other things it does in terms of permissions - eg allowing full screen access without an address bar, but that's irrelevant to this discussion

tablatronix, 2018-07-14:
I will try on my iphone tomorrow , but I am pretty sure the captive portal is sandboxed fairly well, as in there is no way to do any of that.

closed 2023-09-08 by tablatronix