tzapu/WiFiManager #659: setConfigPortalBlocking and additional parameter portail not compatible

spilz87, 2018-07-15:
I'm not able to use setConfigPortalBlocking(false) and add parameters on portail on the same project, I tried to merge the 2 exemples and each time I got a reset loop 

Some one as succed on it or is it an issue ?

tablatronix, 2018-07-18:
is this still an issue?

spilz87, 2018-07-18:
For me yes it is, I'm not able to merge the both exemples

tablatronix, 2018-07-18:
do you have the latest commits?

spilz87, 2018-07-18:
I download the last development branch last week end

tablatronix, 2018-07-18:
It has been updated since then

spilz87, 2018-07-18:
Ok sorry 
I will download the new one and try it again

spilz87, 2018-07-18:
I will let you know

irwanayeah, 2018-08-29:
hello @ spilz87 @tablatronix , is there a solution? because I got the same thing

I use the sample WifiManager \ examples \ DEV \ OnDemandConfigPortal \ OnDemandConfigPortal.ino

then add lines
setConfigPortalBlocking (false); on setup
process (); on the loop

parameters never seen.

tablatronix, 2018-08-29:
Your parameters are setup in setup then they are out of scope in loop. You have to put then somewhere else or make them static or something

irwanayeah, 2018-08-29:
work perfect. thank @tablatronix

tablatronix, 2018-08-29:
Yeah that is one of the problems with the non blocking  and has to be documented and a new example made for it, unless we can make it better, like store parm objects in wm.
Then there is the issue of cleaning up when its global etc.

Ideas welcome

irwanayeah, 2018-08-29:
hi @tablatronix , how about function setSaveConfig Callback(save Config Callback); never fires after saving value of parameters.

tablatronix, 2018-08-29:
Yeah I keep forgetting to work on that. Can you create an issue so I dont forget again.

Since params can now be saved seperate from wifi I need to add new callbacks also

irwanayeah, 2018-08-29:
ok