tzapu/WiFiManager #662: Adding custom header on the main page?

vaz83, 2018-07-16:
#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: ESP8266 (nodeMCU module) 
Hardware:			ESP-12e, esp01, esp25
Core Version:      	2.4.0, staging

### Description

Hi, on the WebPortal of the WiFiManager, under the buttons, there's a header which tells me  if i am connected to an AP and what's the IP address. My question is if can I add a second header (and how)  to inform the user about other stuff (MQTT status, etc) ?

tablatronix, 2018-07-16:
I guess I can add a footer for now.

You can add static html in the strings.h but sounds like you want more dynamic content.

vaz83, 2018-07-16:
Yes, I wanted to have for example:
MQTT Status: Connected / Disconnected
Last Command Received: 0

vaz83, 2018-07-17:
Will it be available in the future ?

tablatronix, 2018-07-17:
Yes I am still thinking about it

vaz83, 2018-07-20:
I've changed the cpp, h, and strings file. Already working here :)

vaz83, 2018-07-20:
Tablatronix I can help you guys with code as well, if you need of course

tablatronix, 2018-07-20:
I am trying to determine if I should add just an html_footer or a way to add stuff inside the callout ( thats the information box on the bottom )

vaz83, 2018-07-20:
I’ve changed the code on the handle root, to display the information I need. Then made a void that receives the data I want and display on the main menu

tablatronix, 2018-07-20:
Where did you add it, what was a good location ? at the bottom ? before or after the status

You made a "function" (void) is the return value of that function

vaz83, 2018-07-21:
[image]

I am sending you the print screen, i think it's the best option. I have also added changing colors from green to red when device is connected/disconected. 
i added a void WiFiManager::setMQTTStatus(bool connected) and a void WiFiManager::showMQTTStatus(bool show) on the cpp file. When I set the showMQTTStatus to true, it reads the flag sent on the setMQTTStatus. To know which ip is the MQTT, i called the getParamOut and read only the parameters that I needed.
I have also made the following modifications:
=> added a back button on the configure wifi page, and the info page
=> Changed the code when I access the webportal, so that the title is the devicename and not the IP

zen85, 2020-05-16:
> [image]
>
> I am sending you the print screen, i think it's the best option. I have also added changing colors from green to red when device is connected/disconected.
> i added a void [...]

hmm... very neat. is this somewhere implemented? i can't find it :(

tablatronix, 2020-05-16:
I think a callback for page header and footer is good

vaz83, 2020-05-17:
> hmm... very neat. is this somewhere implemented? i can't find it :(

I made it... i have added many many options meanwhile..

tablatronix, 2020-05-17:
Did you use the param child class to add those features?

stzoran1, 2023-01-28:
You can do it now using following method:
`wifiManager.setTitle("YourTitle");`