tzapu/WiFiManager #663: Changing template.h

vaz83, 2018-07-17:
Hi,
I am making some changes on the template.h file (changing the background color of the webportal), but everytime i compile, and then I open the page on the webbrowser, it's not changing the color, I view the code, and my changes are not there, seems the compiler is not noticing the changes I have made. Am i doing something wrong ?

Thanks!

vaz83, 2018-07-18:
Sorry, my mistake, i meant using the strings_en.h. I have added in the code #define WIFI_MANAGER_OVERRIDE_STRINGS and added the consts i wanted to change before the #include <WiFiManager.h> but it isn't changing anything, only if i change the strings_en.h directly..

tablatronix, 2018-07-18:
Yeah you cannot define in sketch, I have not figured out how you can override that yet. I think you have to use a shared header. For now you have to modify the strings file.

vaz83, 2018-07-18:
ok, i can live with that :) done it and already working, thanks :)

closed 2018-07-18 by vaz83