tzapu/WiFiManager #670: Compillation error with the git esp8266 Core 2.5.0-dev

jgrulich, 2018-07-22:
When used the last ESP git core 2.5.0-dev, the following build error appears:

> libraries\WiFiManager-master\WiFiManager.cpp.o: In function `WiFiManager::waitForConnectResult()':

D:\arduino-1.8.5\projects\libraries\WiFiManager-master/WiFiManager.cpp:759: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

libraries\WiFiManager-master\WiFiManager.cpp.o: In function `String::operator+=(__FlashStringHelper const*)':

D:\arduino-1.8.5\projects\libraries\WiFiManager-master/WiFiManager.cpp:759: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

collect2.exe: error: ld returned 1 exit status

tablatronix, 2018-07-22:
#645

tablatronix, 2018-07-22:
704a7ae393e8b8f820a73924d0480aae85781c26

closed 2018-07-22 by tablatronix