tzapu/WiFiManager #677: Scan sometimes crashes

martinberlin, 2018-07-25:
Serial Output:
``
*WM: SET AP STA
DISPLAY prints: Display can't connect to WiFi. Entering config mode
Please connect to: Display-autoconnect Access Point
And browse 192.168.4.1 to configure the display WiFi connection.
192.168.4.1
Display-autoconnect
*WM: 
*WM: Configuring access point... 
*WM: Display-autoconnect
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

Exception (29):
epc1=0x40214fe4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

``
Question: Using mDNS will somehow affect wifiManager ? I guess not, but I just want to make sure.

### Basic Infos
WemosD1 ESP8266
Sketch uses 366329 bytes (35%) of program storage space. Maximum is 1023984 bytes.
Global variables use 68928 bytes (84%) of dynamic memory

#### Hardware
WiFimanager Branch/Release:  master

Esp8266/Esp32: Esp8266

Hardware:			ESP-12e

### Description

Tried using development branch but there I see no scan. All other options work correctly, so I can use Sit, just curious to see if I can get the scan to work.

Sketch to reproduce the failure is here:
https://github.com/martinberlin/esp/tree/implement/wifimanager/e-paper-7.5-Skygate

Scan quite randomly crashes or the HTML is rendered partially (For ex. List but no Form below, or list+Form but no Save button)

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 
#include <ESP8266mDNS.h>
// display libs
#include <SPI.h>
#include <GxEPD.h>
#include <GxGDEW075T8/GxGDEW075T8.cpp>
#include <GxIO/GxIO_SPI/GxIO_SPI.cpp>
#include <GxIO/GxIO.cpp>

### Sketch
[image]

https://github.com/martinberlin/esp/tree/implement/wifimanager/e-paper-7.5-Skygate

### Additional info

Partial rendering sometimes:

[image]

```

tablatronix, 2018-07-26:
try development? shrug

martinberlin, 2018-07-26:
Right hint! Thank you very much.
Installing the ESP libraries from here: https://github.com/esp8266/Arduino#using-git-version 
Also removed this two libraries leaving alone WiFiManager.h

    //#include <DNSServer.h>
    //#include <WiFiClient.h>
    #include <WiFiManager.h>

Worked and now seems to be stable. But the only part I could not solve is that Wifi scan page is still rendering 1 time in each 5 halfways like I described before. Did anyone had also this ?

tablatronix, 2018-07-26:
Could be timing out somehow

martinberlin, 2018-07-27:
Thanks tablatronix. And how is possible to solve the time out ? It works better now but it’s sgtill not stable enough.
More info about the chip: 
Chip ID
5931965
Flash Chip ID
1458280
Flash Size
4194304 bytes
Real Flash Size
4194304 bytes
SDK Version
2.2.1(cfd48f3)
Core Version
9c846bd5
Boot Version
31
CPU Frequency
80MHz
Memory - Free Heap
2704 bytes available
Memory - Sketch Size
Used / Total bytes
366384 / 3143472

martinberlin, 2018-07-29:
I solved it editing menu and disabled scan. However would like to discover what it is! 
One thing I noticed is that fails more when I scan a place that is full of WiFis so can be very well a time-out. What I don’t know is what configuration to change to make it work.

tablatronix, 2018-07-29:
Not sure I have over 20 aps in my scans, maybe one of the aps has a weird name and crashes it?

tablatronix, 2018-07-29:
Hmm i noticed you have some quality skips, i wonder if that has something to do with reproducing this.

martinberlin, 2018-07-30:
>Not sure I have over 20 aps in my scans, maybe one of the aps has a weird name and crashes it?

Can be but it's hard to see what name can be causing the thing since it comes after the whole scan it's not stop in the middle. Actually it comes at the end, when in debug I see *WM: Sent config page -> Also when it's sending the HTML is the moment that crashes. That's why sometimes the scan page is rendered before it ends.

>Hmm i noticed you have some quality skips

Yes! I tried to get them out but it makes no difference. Actually without scan it works very stable stable

tablatronix, 2018-07-30:
Can you try turning this off?

`    boolean       _preloadwifiscan        = true;  // preload wifiscan if true` in .h file

martinberlin, 2018-07-31:
I tried but it didn't make any difference. The only way to safely run it in my case is controlling the menu and taking away the scan function. This is what makes it crash on my Wemos D1.
I would be happy to help to debug this one in any way I possibly can :)

tablatronix, 2018-07-31:
Do you always get an exception 29? When this happens and does it happen repeatedly or sometimes work.

martinberlin, 2018-08-03:
It sometimes work, but only 1 of each 10 times or so, comes mostly this exception and sometimes no exception but the HTML is rendered only half ways like described before. 
I left a link on the top to my sketch, maybe I'm doing something that is coming on the way, but curiously without this scan stuff it works very well and it's stable enough for me. 
Scan would be nice, but it's not a must in this case, since I build stuff we use internally at the office. So I use it mostly when I come home and make some update. Thanks anyways for the support!

tablatronix, 2018-08-04:
Youll probably have to decode that exception.
Theres a lot going on in that sketch

martinberlin, 2018-08-10:
Thanks. I will try this weekend to use this 
https://github.com/me-no-dev/EspExceptionDecoder Restart the scan and check what is going on. Thanks!

tablatronix, 2018-08-10:
What esp lib again, have you tried staging?
Others have also reported this error.

martinberlin, 2018-08-18:
I'm on the last updated master of the Arduino esp8266 library on github. Is there any other branch that I should switch ?   
https://github.com/esp8266/Arduino -> I don't see any staging branch there

tablatronix, 2018-08-18:
that is staging

martinberlin, 2018-08-20:
I just wanted to comment that in another projects like for example a [digital security camera](https://github.com/martinberlin/esp/tree/master/FasarekCam) I'm making with a simple Arducam this is not happening. Also there must be something not playing well together with the GxEPD e-ink display library and your WifiManager. 
Since for other things it works good and it does not trigger this bug.

tablatronix, 2018-08-20:
Well if you guys get a chance , decode your exceptions and we can figure it out, ill bet its a esp lib bug. Or a memory issue. 

What chips are you using, make sure to run a flash check sketch to make sure your flash is ok, and do a fulll flash erase.

martinberlin, 2018-11-07:
I can confirm I didn't get this issue anymore on any of my sketches. That was a project with a big E-ink display and it was the only one where I got this issue. Now I upgraded to the latest core and to develop of your library and it's not coming anymore. So I bet it was not your bug but as you mention in other layers. Thank you very much for replying and your support!

tablatronix, 2018-11-07:
interesting I forgot about this issue, I wonder what it was, I probably need to to make sure there are no memory leaks still just in case.

closed 2018-11-07 by tablatronix