tzapu/WiFiManager #679: Master: Sometimes crash after User requested captive portal

dontsovcmc, 2018-07-27:
I user master branch for stability, but sometimes got loop exceptions when boot, read smth by i2c and start Wi-Fi connecting.

I can't understand the reason. I check turn on 2 times after firmwaring... 

I know, that I should use development branch. My first experience got an error (with buffer lenght) and I return to master branch. Maybe I can help with development?

Exception decoder:
```
Decoding 42 results
0x401004f4: calloc at ?? line ?
0x40106b98: pvPortZalloc at ?? line ?
0x4010577c: spi_flash_read at ?? line ?
0x40236131: wifi_param_save_protect_with_check at ?? line ?
0x401063f1: __wrap_spi_flash_read at ?? line ?
0x4023621b: system_param_save_with_protect at ?? line ?
0x402361fe: system_param_save_with_protect at ?? line ?
0x402366e8: wifi_station_ap_number_set at ?? line ?
0x401063f1: __wrap_spi_flash_read at ?? line ?
0x40236902: wifi_station_ap_number_set at ?? line ?
0x402369ba: wifi_station_ap_number_set at ?? line ?
0x4024d990: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40236989: wifi_station_ap_number_set at ?? line ?
0x402369f0: wifi_station_set_config at ?? line ?
0x40206cb4: ESP8266WiFiSTAClass::disconnect(bool) at ?? line ?
0x40213b00: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x4020febc: _printf_i at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 241
0x40213f60: _svfprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 641
0x4020ee94: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020a311: Print::write(char const*) at ?? line ?
0x4020a324: Print::print(char const*) at ?? line ?
0x40205a02: setup_ap(Settings&, SlaveData const&, float const&, float const&) at ?? line ?
0x4020a311: Print::write(char const*) at ?? line ?
0x4020ee94: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020a311: Print::write(char const*) at ?? line ?
0x4020a44d: Print::printNumber(unsigned long, unsigned char) at ?? line ?
0x40210624: sprintf at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/sprintf.c line 646
0x4020ee94: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020ee94: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020a4ab: Print::print(long, int) at ?? line ?
0x4020ee94: Print::write(unsigned char const*, unsigned int) at ?? line ?
0x4020a311: Print::write(char const*) at ?? line ?
0x4020a324: Print::print(char const*) at ?? line ?
0x40240000: etharp_tmr at netif/etharp.c line 232
0x40202ea0: calculate_values(Settings&, SlaveData&, float*, float*) at ?? line ?
0x4020a50e: Print::print(unsigned long, int) at ?? line ?
0x4020469c: loop at ?? line ?
0x40209fb4: EspClass::wdtDisable() at ?? line ?
0x40202bda: setup at ?? line ?
0x4020b420: loop_wrapper() at core_esp8266_main.cpp line ?
0x40100700: cont_norm at cont.S.o line ?
```

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master
Esp8266-01

Core Version:
Platform: Espressif 8266 1.7.3

### Settings in IDE
PlatformIO:
[env:esp01_1m]
board = esp01_1m
platform = espressif8266
framework = arduino

### Debug Messages

```
ets Jan  8 2013,rst cause:2, boot mode:(3,5)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
000:00:00:00:050  NOTICE    (ESP) : Booted
1 000:00:00:00:053  NOTICE    (I2C) : mode=1
4 1 E4 B 0 0 9 0 0 0 31 0 0 0 000:00:00:00:065  NOTICE    (I2C) : version: 4
000:00:00:00:065  NOTICE    (I2C) : service: 1
000:00:00:00:065  NOTICE    (I2C) : voltage: 3044
000:00:00:00:067  NOTICE    (I2C) : impulses0: 9
000:00:00:00:071  NOTICE    (I2C) : impulses1: 49
000:00:00:00:075  NOTICE    (WIF) : key=51a655b2819f4d11ae784e28d83651e4 email=, hostname=blynk-cloud.com
000:00:00:00:085  NOTICE    (WIF) : value0_start=0.06, impules0_start=5, factor=10
000:00:00:00:092  NOTICE    (WIF) : value1_start=0.45, impules1_start=44
000:00:00:00:098  NOTICE    (ESP) : new impulses=9 49
000:00:00:00:103  NOTICE    (ESP) : new values=0.10 0.50
000:00:00:00:108  NOTICE    (ESP) : I2C-begined: mode SETUP
000:00:00:00:113  NOTICE    (WIF) : User requested captive portal

Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006ed31 depc=0x00000000

ctx: cont 
sp: 3fff0740 end: 3fff0cc0 offset: 01a0

```

tablatronix, 2018-07-27:
Shrug, maybe someone will see this and help.
But no fixes or dev will be done on master, hotfixes has some fixes.

Are you using parameters?

Have you erased flash.

Have you tried esp staging?

tablatronix, 2018-07-27:
Could be out of memory, I have not done any testing on esp01 at all personallly

tablatronix, 2018-07-27:
What esp lib version are you using?
I doubt 1.7.3 that would be 3 years old

dontsovcmc, 2018-07-28:
I have a lot of parameters: 

I mean platformio platform version: https://github.com/platformio/platform-espressif8266
This is [compile options](https://github.com/platformio/platform-espressif8266/blob/develop/boards/esp01_1m.json)
	
```
#define KEY_LEN 34
#define HOSTNAME_LEN 32

#define EMAIL_LEN 32
#define EMAIL_TITLE_LEN 64
#define EMAIL_TEMPLATE_LEN 200
...
	WiFiManagerParameter param_key( "key", "key",  sett.key, KEY_LEN );
	wifiManager.addParameter( &param_key );
	WiFiManagerParameter param_hostname( "host", "host",  sett.hostname, HOSTNAME_LEN );
	wifiManager.addParameter( &param_hostname );
	WiFiManagerParameter param_email( "email", "email",  sett.email, EMAIL_LEN );
	wifiManager.addParameter( &param_email );
	WiFiManagerParameter param_email_title( "title", "title",  sett.email_title, EMAIL_TITLE_LEN );
	wifiManager.addParameter( &param_email_title );
	WiFiManagerParameter param_email_template( "template", "template",  sett.email_template, EMAIL_TEMPLATE_LEN );
	wifiManager.addParameter( &param_email_template );
	FloatParameter param_value0_start( "value0", "value0",  value0);
	wifiManager.addParameter( &param_value0_start );
	FloatParameter param_value1_start( "value1", "value1",  value1);
	wifiManager.addParameter( &param_value1_start );
	LongParameter param_litres_per_imp( "factor", "factor",  sett.liters_per_impuls);
	wifiManager.addParameter( &param_litres_per_imp );
```

liebman, 2018-07-28:
If you declare a lot of parameters in a function (on the stack) you will exceed the stack size and memory will be corrupted.  You may need to use the `new` operator for the allocation and then the `delete` operator to free them.  (I had this issue with 31 parameters)

liebman, 2018-07-28:
If I recall correctly the default stack size is only 4k. I don't recall whats adjacent to it, I think its the heap. So it would depend on if that particular memory location is in use or not.

dontsovcmc, 2018-08-04:
WiFimanager Branch/Release: Master

I think there is no problem with stack and there is no problem with strings.

Crash when I call:
`WiFi.disconnect( true );`
before run Active Point. 

But I don't know why I called this function... 
I read, that it erases current SSID and password. I don't need it.

I see example [OnDemandConfigPortal](https://github.com/tzapu/WiFiManager/blob/development/examples/OnDemandConfigPortal/OnDemandConfigPortal.ino) without any additional functions.

closed 2018-08-04 by dontsovcmc

tablatronix, 2018-08-04:
Known esp bug update your library