tzapu/WiFiManager #682: AP mode wifi configuration html page is not getting opened!!!

chandan-skillfinity, 2018-08-02:
Hardware:		NodeMCU 1.0(ESP-12e Module)
Core Version:      	2.4.1
Arduino IDE:           1.8.5
Operating System: MacOS High Sierra 10.13.6 

Following 3 issues observed, request help in solving 
1. After going to AP mode, WiFi Manager not opening the configuration page.
2. Sometimes the program also crashing
3. Once mobile connects to AP MODE then neither it shows the web page nor it times out (it is stuck inside the method for ever) 

### Sketch
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
WiFiManager wifiManager;

void setup()
{
    Serial.begin(38400);

    Serial.println("entered setup");

    
    wifiManager.setTimeout(60);
    
    if(!wifiManager.autoConnect("AutoConnectAP"))
    {
      Serial.println("failed to connect and hit timeout");
      delay(3000);  
    }
    else
    Serial.println("connected");   
}

void loop() 
{
    Serial.println("entered loop");

    if(WiFi.status()== WL_CONNECTED)
    {
       Serial.println("In Loop Connected");
    }
    else
    {
        if(!wifiManager.autoConnect("AutoConnectAP"))
        {
          Serial.println("failed to connect and hit timeout");
          delay(3000);
         }
         else
         Serial.println("In loop connected");
    }
}
```

### Debug Messages

*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
failed to connect and hit timeout
entered loop
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
failed to connect and hit timeout
entered loop
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
failed to connect and hit timeout
entered loop
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: sys 
sp: 3ffffc50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdf0:  00000000 00000000 00000000 4021fad6  
3ffffe00:  4021fba2 4021fac0 3fff12bc 40100ec6  
3ffffe10:  3ffe9815 401042c3 3ffed2a0 00000030  
3ffffe20:  00000018 3ffedc70 3ffedbec 00000000  
3ffffe30:  4021fc63 3ff20a00 3fff12bc 3ffeba23  
3ffffe40:  3ffeb9fc 00000000 00000000 3fff0e54  
3ffffe50:  0000001c 2e9ce79f 401021f2 3ffeb9fc  
3ffffe60:  3ffeba23 0000002c 00000008 401004f4  
3ffffe70:  3ffeba06 3fff0f74 4021c828 3ff20a00  
3ffffe80:  3fff12bc 3fff1e84 00000015 4021c6fb  
3ffffe90:  3fffc278 40101f8c 3fffc200 00000022  
3ffffea0:  3ffea0b4 4021f3c7 3ffedbec 2e9cb4ed  
3ffffeb0:  40000f68 00000030 00000010 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  0000009a 3fff1fb8 00000000 ffefffff  
3ffffee0:  ffffffff 3fffc6fc 00000000 3fffdab0  
3ffffef0:  3ffeba14 3fffdc10 00000031 00000030  
3fffff00:  402244cd ffffffd1 00000000 3ffea7b1  
3fffff10:  4021bf37 3ffeb9fc 3ffed2a0 3fff12bc  
3fffff20:  00000000 3ffeba06 3ffeba29 00000000  
3fffff30:  402321f1 00000000 00000000 3ffeba06  
3fffff40:  4023139c 00000047 3ffeb9f0 3fff0f74  
3fffff50:  3ffeb9f0 00000000 00000031 40230cc6  
3fffff60:  3ffed2a0 3ffeb9f0 3fffdcc0 3ffe98d0  
3fffff70:  3ffeba0c 3fff0e54 00000040 3ffe98d0  
3fffff80:  00000040 3ffed2a0 00000002 40106e7c  
3fffff90:  402305a7 3fffdab0 00000000 4023b9d6  
3fffffa0:  3ffe98d0 40000f49 3fffdab0 40000f49  
<<<stack<<<
@⸮ *⸮⸮⸮*⸮D⸮@⸮ *⸮ԕ⸮⸮*⸮D⸮⸮

```

tablatronix, 2018-08-02:
This is development branch?

chandan-skillfinity, 2018-08-03:
No, we downloaded the WiFi manager from the released branch

tablatronix, 2018-08-03:
not sure, sounds like a memory problem or sdk problem, try development, issues against master are ignored