tzapu/WiFiManager #684: Two small problems

Mohammed-Kher-Ali, 2018-08-03:
[image]

1-The top line disappears in the text input box. How can I format this better
2-It works by resetting and configuring an access point after the power outage because ESP 8266  works faster than the internet router where you are searching ESP 8266 so the network does not do a reset job while the router needs 30 seconds to 1 minute to start broadcasting the internet. Can this be delayed for a minute In the absence of a network

tablatronix, 2018-08-03:
setconnecttimeout

Mohammed-Kher-Ali, 2018-08-03:
i have fixed .issue 1
Keep adding "setconnecttimeout

" to my code

```

#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <EEPROM.h>
#include <WidgetRTC.h>

WidgetRTC rtc;
BlynkTimer Time;

    WiFiManager wifiManager;
    WiFiClient wifiClient;

#define EEPROM_SALT 12664
typedef struct
  {
  int   salt = EEPROM_SALT;
  char Blynk_Token[33]  = "";
  char Device_Name[20]  = "Smart Mini Relay";
  char Widgets_Color[8]  = "#23C48E";
  char Pass_Setting[9]  = "12345";
  }

WMSettings;
WMSettings blynk;
WMSettings device;
WMSettings color;
WMSettings pass;

void eeprom_read()
  {
  EEPROM.begin(512);
  EEPROM.get(0, blynk);
  EEPROM.get(100, device);
  EEPROM.get(200, color);
  EEPROM.get(300, pass);
  EEPROM.end();
  }

void eeprom_saveconfig()
  {
  EEPROM.begin(512);
  EEPROM.put(0, blynk);
  EEPROM.put(100, device);
  EEPROM.put(200, color);
  EEPROM.put(300, pass);
  EEPROM.commit();
  EEPROM.end();
  }

/*********************************************************************************/

void configModeCallback (WiFiManager *myWiFiManager)
  {
  String ssid =  device.Device_Name;
  }

void setup()
  {
  Serial.begin(115200);
  pinMode(Relay, OUTPUT);

 

  eeprom_read();

  if (blynk.salt != EEPROM_SALT)
    {
    Serial.println("Invalid settings in EEPROM, trying with defaults");
    WMSettings defaults;
    blynk = defaults;
    device = defaults;
    color = defaults;
    pass = defaults;
    }

  String ssid =  device.Device_Name;
  WiFi.hostname(ssid);

  
  WiFiManagerParameter custom_blynk_text("Device ID");
  wifiManager.addParameter(&custom_blynk_text);

  WiFiManagerParameter custom_blynk_token("blynk-token", "Blynk Token", blynk.Blynk_Token, 33);
  wifiManager.addParameter(&custom_blynk_token);

  WiFiManagerParameter custom_text("<p>This is just a text paragraph</p>");
wifiManager.addParameter(&custom_text);

  WiFiManagerParameter custom_device_text("Device Name");
  wifiManager.addParameter(&custom_device_text);

  WiFiManagerParameter custom_device_token("device-token", "Device Name",  device.Device_Name, 20);
  wifiManager.addParameter(&custom_device_token);

  WiFiManagerParameter custom_color_text("Widget Color");
  wifiManager.addParameter(&custom_color_text);

  WiFiManagerParameter custom_color_token("color-token", "Hex Color eg.#FF0000 = Red color", color.Widgets_Color, 8);
  wifiManager.addParameter(&custom_color_token);

  WiFiManagerParameter custom_pass_text("Lock settings");
  wifiManager.addParameter(&custom_pass_text);

  WiFiManagerParameter custom_pass_token("pass-token", "eg.123Abc", pass.Pass_Setting, 9);
  wifiManager.addParameter(&custom_pass_token);

  if (!wifiManager.autoConnect(ssid.c_str(), NULL)) 
    {
    Serial.println("failed to connect, we should reset and see if it connects");
    delay(6000);
    WiFi.disconnect();
    wifiManager.resetSettings();
    ESP.reset();   
    }

  strcpy(blynk.Blynk_Token, custom_blynk_token.getValue());
  strcpy(device.Device_Name, custom_device_token.getValue());
  strcpy(color.Widgets_Color, custom_color_token.getValue());
  strcpy(pass.Pass_Setting, custom_pass_token.getValue());
 
  eeprom_saveconfig();
  
  
  Blynk.config(blynk.Blynk_Token);
  bool result = Blynk.connect();
  delay(1000);

  if (result != true)
    {
    Blynk.config(blynk.Blynk_Token);
    Serial.println("BLYNK Connection Fail");
    delay(1000);
    wifiManager.resetSettings();
    ESP.reset();
    }
  else
    {
    Serial.println("BLYNK Connected");

    }

  }

void loop()
  {
  Blynk.run();
  Time.run();
  }

```

rodri16, 2018-08-08:
How did you solve problem 1?

Mohammed-Kher-Ali, 2018-08-08:
[image]
[image]

I'll post the edit when you're finished

rodri16, 2018-08-09:
Did you change only strings_en.h or something else?

Mohammed-Kher-Ali, 2018-08-09:
It is not connected to the Internet in this case so I think it is difficult to add a picture

mikekgr, 2018-08-09:
Dear @Mohammed-Kher-Ali 
do you have the above code but full, somewhere published ? GitHub perhaps?
If yes, can you share the URL?
Thanks

Mohammed-Kher-Ali, 2018-08-09:
Icon I modified it
Editing not yet finished
Modified for style only
I'll post it when you're finished

tablatronix, 2018-08-09:
you can customize the menu but not add stuff, unless you edit the strings file manually,

You can add whatever you want, mohammed seems to be still using the master branch.

The development branch even support invert dark theme

Mohammed-Kher-Ali, 2018-08-14:
#https://github.com/Mohammed-Kher-Ali/WiFi-Manager

[image]
[image]
[image]
[image]
[image]
[image]
[image]

tzapu, 2018-08-14:
that looks cool :D

closed 2018-08-16 by tablatronix