tzapu/WiFiManager #686: How to use WiFi Manager without the captive portal?

sabersingh, 2018-08-04:
Hi there,
I want to use the WiFi manager to change the wifi credentials of my ESP device but without using the Captive portal, I want the user to configure the WiFi settings either from my app or from a web interface on the mDNS address(I don't want the captive portal to pop-up). If I can change the wifi credentials only by using the HTTP request then It should supposedl work for me.
**PLEASE SOMEBODY HELP ME OUT ASAP, I HAVE A PROJECT TO SUBMIT. PLEASE!**
Thanks in advance

sabersingh, 2018-08-04:
Hi @tablatronix 
Thanks for such a quick reply. Actually I am currently in a very depressed state of mind and this project is very important for me, I won't be able to go through the documentations again so please it would be of great help if you could put some light on it and briefly explain how exactly is that supposed to be done.
Thank you so very much once again for understanding.

sabersingh, 2018-08-04:
Also is there a way to just remove the web interface and keep features of WiFi manager limited to HTTP request?

sabersingh, 2018-08-07:
I just want to connect to/Change WiFi credentials at runtime using my app or from a touch display that I have on board my Device, but I don't want Web interface/ Captive portal to show up. Just use the HTTP requests and connect to the WiFi

tablatronix, 2018-08-08:
```cpp
    //if false, disable captive portal redirection
    void          setCaptivePortalEnable(boolean enabled);
```
development

sabersingh, 2018-08-08:
Oh! thanks Shawn, That was pretty neat! also what about If I want to change the WiFi credentials during runtime only by using the hardware side, In simple words, I have a TFT display using which I want to change the WiFi credentials programatically, is it possible?

tablatronix, 2018-08-08:
I see no need to use wifimanager for that, just use esp begin(ssid,pass)

sabersingh, 2018-08-08:
I tried using esp begin function but it only works for the very first time when I reset the device but it wont work during the runtime.

tablatronix, 2018-08-08:
you might have to disconnect first

sabersingh, 2018-08-08:
Oh I see.. can you please share some example?

tablatronix, 2018-08-08:
WiFi.disconnect();

sabersingh, 2018-08-08:
Thank you so very much @tablatronix for helping and for all the assistance.