tzapu/WiFiManager #688: DNS lookups fail if config portal was up.

liebman, 2018-08-06:
### Basic Infos

Development branch (current development branch git)

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32:  esp8266

Hardware:			ESP-12e

Core Version:      	current master from git

### Description

DNS name resolution fails if config portal was up.  When autoconnect without portal DNS name resolution works fine.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: None

### Sketch

```cpp
#include <Arduino.h>
#include <WiFiManager.h>

const uint8_t PIN_TRGR = 14;
const char* AP_NAME = "DNSFail";
const char* server_name = "0.zoddotcom.pool.ntp.org";

void setup() {
    Serial.begin(76800);

    WiFiManager wm;
    
    if (digitalRead(PIN_TRGR) == 0)
    {
        wm.startConfigPortal(AP_NAME, NULL);
    }
    else
    {
        wm.autoConnect(AP_NAME);
    }
    

    Serial.println("connected...yeey :)");
}

void loop() {
    Serial.printf("lookup '%s' ... ", server_name);
    IPAddress address;
    if (!WiFi.hostByName(server_name, address))
    {
        Serial.println("FAILED!");
    }
    else
    {
        Serial.printf("%d.%d.%d.%d\n", address[0], address[1], address[2], address[3]);
    }

    delay(1000);
}
```

### Debug Messages

Reset with pin 14 LOW - force config portal - DNS lookups fail.
[... 318 lines cut ...]
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
wifi evt: 7
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 173.255.215.209
173.255.215.209

```

tablatronix, 2018-08-06:
Cannot reproduce

```
SDK Version
2.2.1(cfd48f3)
Core Version
2_4_1
```

tablatronix, 2018-08-06:
Can you add a     `Serial.println(wm.getModeString(WiFi.getMode()));` to make sure you are connected to sta when coming out of configportal, also does using configportaltimeout (and letting it timeout NOT saveing ) result in the same behavior ?

tablatronix, 2018-08-06:
only this I could find with this offer_error is this
https://github.com/nodemcu/nodemcu-firmware/pull/1985

liebman, 2018-08-06:
I'll give the above print a try when I get home.

tablatronix, 2018-08-06:
ok let me enable debugging and see if mine does also , I think that happens when you restart sta or change modes. Ill see if its related

tablatronix, 2018-08-06:
I tried it with sta failing and saved, both work, although I did get 1 error, probably a timing issue,

```
*WM: [1] config portal exiting 
STA+AP
connected...yeey :)
*WM: [3] unloading 
lookup 'example.com' ... [hostByName] request IP for: mtsu.edu
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 6
[hostByName] Host: example.com lookup error: -5!
FAILED!
lookup 'example.com' ... [hostByName] request IP for: mtsu.edu
[hostByName] Host: example.com IP: 10.14.0.116
10.14.0.116
```

liebman, 2018-08-06:
I'm getting "STA+AP" - even if I add `WiFi.mode(WIFI_STA);` before starting the config portal.

Adding a 30 second timeout yields this where it fails a few times then says its connected and starts working.

```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

SDK:2.2.1(cfd48f3)/Core:unspecified/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
*WM: [3] WIFI station disconnect 
scandone
*WM: [3] WiFi station enable 
del if0
usl
mode : null
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  DNSFail
modwifi evt: 8
e : softAP(6a:c6:3a:ac:8b:4e)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 8
*WM: [2] AP has anonymous access! 
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
mode : sta(68:c6:3a:ac:8b:4e) + softAP(6a:c6:3a:ac:8b:4e)
add if0
wifi evt: 8
scandone
*WM: [2] WiFi Scan done in 2198ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: [2] Portal Timeout In 0 seconds
*WM: [1] config portal has timed out 
*WM: [3] configportal abort 
*WM: [2] disconnect configportal 
bcn 0
del if1
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [1] config portal exiting 
STA+AP
connected...yeey :)
*WM: [3] unloading 
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org lookup error: -6!
FAILED!
wifi evt: 2
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org lookup error: -6!
FAILED!
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org lookup error: -6!
FAILED!
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with ThicketLand, channel 6
dhcp client start...
wifi evt: 0
ip:192.168.0.30,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 3
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 162.248.241.94
162.248.241.94
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 162.248.241.94
162.248.241.94
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 162.248.241.94
162.248.241.94
lookup '0.zoddotcom.pool.ntp.org' ... [hostByName] request IP for: 0.zoddotcom.pool.ntp.org
[hostByName] Host: 0.zoddotcom.pool.ntp.org IP: 162.248.241.94
162.248.241.94
```

liebman, 2018-08-06:
And adding `WiFi.mode(WIFI_STA);` after the return from config portal also works.

tablatronix, 2018-08-07:
Can you test one more thing?
Add a 2 second delay before when starting configportal to give esp time to connect, autoconnect=true?

tablatronix, 2018-08-07:
Thats because there was a bug and I had to remove this
```  // WiFi_Mode(_usermode); // restore users wifi mode, BUG https://github.com/esp8266/Arduino/issues/4372```

tablatronix, 2018-08-07:
How did you output this ?

```
SDK:2.2.1(cfd48f3)/Core:unspecified/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
```

tablatronix, 2018-08-07:
Also it looks like your lookup failures have changed to a different error.. hmm

liebman, 2018-08-07:
I'm using Debug port: Serial & Debug Level: WiFi

I wonder if the DNS issue is because AP is enabled and somehow the DNS lookup is being set there (and fails of course).   https://github.com/esp8266/Arduino/issues/4949 mentions this possibility in one of the comments.  Would explain why it works if I set STA after the config portal exits.

tablatronix, 2018-08-07:
Maybe, But I was in not seeing issues with the same mode. Shouldn't dns work on both interfaces?
Or are we breaking it when we do `  dnsServer.reset();`

tablatronix, 2018-08-07:
ok so if the subnets are the same, the loopback dns will be hit it seems.

Is this what you are seeing, is your sta ap dns the same subnet as your softap ?

liebman, 2018-08-07:
Nope - 192.168.0.0/24 vs 192.168.4.0/24  Maybe there is something odd about how my router sends teh DNS server? (But it works in STA only mode)

tablatronix, 2018-08-07:
strange I will try to see of that bug can be fixed for now, if that issue is still a problem or not