tzapu/WiFiManager #689: Problems with custome parameters after restart

skywalker1979, 2018-08-06:
### Basic Infos
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (29)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Alice, channel 11
dhcp client start...
ip:192.168.3.197,mask:255.255.255.0,gw:192.168.3.1
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.3.197
connected...yeey :)
local ip
192.168.3.197
*WM: freeing allocated params!

Latest official relelease, but also tried latest development of WifiManager

Hardware:			esp01

Core Version:      	2.4.2

Problem description
I flashed the sample "AutoConnectWithFSParameters" and made no changes. (except adding a line in loop, that would tell me if the custom parameter has been read.)
Settings are saved as soon as they are entered, but are lost on power off / power on..

### Settings in IDE
Inside arduino i tried Generic esp8266. 
Nodemcu seems not to be working for me.

skywalker1979, 2018-08-18:
i don't know. How can i know if mine supports spiffs?
i don't see any MOUNTING fs or the rest.

skywalker1979, 2018-08-18:
In case spiffs is not supported...is there any chance to use eeprom?

skywalker1979, 2018-08-18:
maybe it is connected to this?

https://github.com/esp8266/Arduino/issues/4061

skywalker1979, 2018-08-19:
it is true that my esp01 have the puya chip....so it seems totally connected.

skywalker1979, 2018-08-23:
i can confirm that the path indicated in the issue i linked, does work and solves the issue with the puya chip not being able to use spiffs....
i hope it will be integrated in the main esp release.

tablatronix, 2018-08-23:
refs
https://github.com/esp8266/Arduino/issues/4061#issuecomment-368273656

tablatronix, 2018-08-23:
if anyone wants to find a replacement for spiffs for those examples feel free to PR against development branch