tzapu/WiFiManager #690: Custom design

nelfata, 2018-08-07:
Hello tzapu,
I need to have some customization made on the portal page, can we discuss this offline?
Please send me an email to: nelfata@gmail.com

Thanks.

tablatronix, 2018-08-07:
you can try the devleopment branch, it makes customization easier, all strings are in strings_en.h

nelfata, 2018-08-08:
I need to have a customized web interface with additional functionality to the captive page. I am not well versed in HTML and Javascript. I will be willing to pay for this effort.
Thanks.

diegoalbaga, 2018-09-18:
How do I ise the development branch??

dontsovcmc, 2018-09-20:
> How do I ise the development branch??
clone repository with git and call 'git checkout development' to choose development branch.

tablatronix, 2018-09-20:
or change branch and click download and put files where you need them

diegoalbaga, 2018-09-22:
Thanks I have already found and learne how to modify them