tzapu/WiFiManager #696: Question about WIFI_AP_STA

laercionit, 2018-08-10:
First of all congratulations on LIB, really did a great job.

I researched a lot, but unfortunately I did not find any information about it.

I would like to keep the project as WIFI_AP_STA.

Is it possible with WifiManager to keep the STA and AP together?

tablatronix, 2018-08-10:
What exactly do you mean ?

laercionit, 2018-08-10:
I apologize if it was not clear.

I wonder if WifiManager has any option to perform the wifi connection as STA and keep the AP mode open. If you do not have access to the wifi router, you could continue to access AP mode without necessarily having to boot the system.

tablatronix, 2018-08-10:
Yes you can either use ondemandwebportal or ondemandconfigpora with non blocking mode

This is in development branchp

rtek1000, 2018-11-19:
Hello, I also want to congratulate you for the great job, in conjunction with OTA is perfect!

I would like to know if it already has some progress on this because sometimes the location does not have a router, and it is difficult to ask the user to turn their smartphone into an Access Point.

Any predictions for this feature?

rtek1000, 2018-11-19:
I also tested esp-link, it's a great job as well, but I have some issues in making UDP connections, esp-link can provide great ideas, esp-link operates as a configuration page for a router, and can also operate on AP mode.

> The esp-link firmware connects a micro-controller to the internet using an ESP8266 Wifi module. It implements a number of features:
> 
>     -- Transparent bridge between Wifi and serial, useful for debugging or inputting into a uC
>     -- Flash-programming attached Arduino/AVR microcontrollers and LPC800-series and other ARM microcontrollers via Wifi
>     -- Built-in stk500v1 programmer for AVR uC's: program using HTTP upload of hex file
>     -- Outbound REST HTTP requests from the attached micro-controller to the internet
>     -- MQTT client pub/sub from the attached micro-controller to the internet
>     -- Serve custom web pages containing data that is dynamically pulled from the attached uC and that contain buttons and fields that are transmitted to the attached uC (feature not fully ready yet)
> 
> 

https://github.com/jeelabs/esp-link

rtek1000, 2018-11-19:
Hello, actually supports AP, but how to use the configuration page to set the SSID and password to AP mode?

tablatronix, 2018-11-19:
you set it in code, if you want to add custom parameters to allow user to change then you have to add that yourself with custom params and save it then read it and apply it

rtek1000, 2018-11-19:
Thank you, I was able to identify a problem with the ESP-01 modules that I was testing, they are new, but the 25Q80 memories seem to be operating in a dubious way, I do not know if they are fake 25Q80. So I installed a 25Q64 instead of 25Q80, and I get faster and more stable. I will continue to experiment with this library, as well as esp-link, which already offers the OTA system for the Arduino. Anyway, I appreciate your attention.

rtek1000, 2018-11-19:
> wm is more a framework and library than an application, you write your own code to do that stuff

Yes, I agree, thank you.

rtek1000, 2018-11-24:
Hello,

To anyone who might be interested, there is an example code that performs the configuration of the SSID and password for AP mode (with external reset button implementation):

https://github.com/msraynsford/APConfig

tablatronix, 2018-11-24:
its pretty trivial with wm, someone just needs to write it, get param, save to eeprom, read eeprom autoconnect(value,pass);